jkjung-avt/tensorrt_demos #92: How to convert trained tensorflow SSD models to UFF with custom resolution

sudo-wang, 2020-04-17:
@jkjung-avt  Hi, thanks for your work, it is very nice!!!
I follow your tutorial to train the ssd_mobilenet_v1 model with DETRAC datasets, 
then I got the trained model(.pb) and it performs well, so i want to convert trained tensorflow models to UFF and tensorRT.
My custom resolution is 256*256, so i modified INPUT_DIMS = (3, 300, 300) to **(3, 256, 256)** in **build_engine.py**. Finally, I encountered the following errors:

**[TensorRT] INFO: Detected 1 inputs and 2 output network tensors.
python: nmsPlugin.cpp:139: virtual void nvinfer1::plugin::DetectionOutput::configureWithFormat(const nvinfer1::Dims*, int, const nvinfer1::Dims*, int, nvinfer1::DataType, nvinfer1::PluginFormat, int): Assertion `numPriors * numLocClasses * 4 == inputDims[param.inputOrder[0]].d[0]' failed.**

I think the problem is with resolution. By the way, i modified the input_order with various permutations and combinations(such as [1, 0, 2]、[0,1,2].....), It still doesn't work.
**So how can I optimise model for custom resolution ? It works fine with your 300x300 resolution. But can't work with my (3, 256, 256)....** 
Thank you in advance for your help!!

jkjung-avt, 2020-04-17:
When you change input image shape of the SSD model, the output feature map shapes would also change.  You need to figure out the correct output shapes and put them into [featureMapShapes](https://github.com/jkjung-avt/tensorrt_demos/blob/master/ssd/build_engine.py#L108).

sudo-wang, 2020-04-17:
@jkjung-avt 
Thanks for your help!! I have solved this problem.

closed 2020-04-17 by sudo-wang

neethuj95, 2020-06-29:
Hi @jkjung-avt ,

I have trained my ssd-mobilenet-v2 model in 300*300 and was able to convert the model to tensor-rt model using the methods specified in your repo, Thanks to you.

But my doubt is this :  I have trained another ssd-mobilenet-v2 model with 480×272. I want to convert this model to TensorRT model. I modified `featureMapShapes=[19, 10, 5, 3, 2, 1]` as you said in [here](https://github.com/jkjung-avt/tensorrt_demos/issues/92) . But it seems this only works, if height and width are same(ie, if aspect ratio is 1).
Do we have a solution for this ? Can you please help me solve this issue

jkjung-avt, 2020-06-30:
If you use TensorRT C++ API, it is possible to specify different heights and widths using the "PriorBox" plugin (https://github.com/NVIDIA/TensorRT/tree/master/plugin/priorBoxPlugin).

But it doesn't seem possible when we use the "GridAnchor" plugin (https://github.com/NVIDIA/TensorRT/tree/master/plugin/gridAnchorPlugin) in the UFF.  The heights and widths are hard-coded to be the same [here](https://github.com/NVIDIA/TensorRT/blob/master/plugin/gridAnchorPlugin/gridAnchorPlugin.cpp#L464) and [here](https://github.com/NVIDIA/TensorRT/blob/master/plugin/gridAnchorPlugin/gridAnchorPlugin.cpp#L469).

jkjung-avt, 2020-06-30:
Reference: https://github.com/NVIDIA/TensorRT/issues/612 and https://github.com/NVIDIA/TensorRT/issues/247