jkjung-avt/tensorrt_demos #95: input image with other resolution to yolov3

jeansely, 2020-04-18:
Hi,
for other resolutions( for example 320*64) where codes have to change.
i want to segment car plate with yolov3 model. i do not know that where do I change the files trt_yolov3.py  & yolov3.py. 

Many Thanks

jkjung-avt, 2020-04-19:
This seems to be a duplicated issue?  https://github.com/jkjung-avt/tensorrt_demos/issues/85

The following error message shows that your output tensor shape is not as expected: 4 != 10*2*1*1.  So I suspect you've added 1 (too-many) "stride=2" in your config file.

`ValueError: cannot reshape array of size 4 into shape (10,2,1,1)`

jeansely, 2020-04-19:
Hi,
I have attached all the relevant files. Due to the changes made, I still have the following problem:

python3 trt_yolov3.py --model yolov3-tiny-288 --image --filename /home/cv/Downloads/charactersegmentation/5.jpg
Traceback (most recent call last):
  File "trt_yolov3.py", line 96, in <module>
    main()
  File "trt_yolov3.py", line 88, in main
    loop_and_detect(cam, trt_yolov3, conf_th=0.3, vis=vis)
  File "trt_yolov3.py", line 56, in loop_and_detect
    boxes, confs, clss = trt_yolov3.detect(img, conf_th)
  File "/home/cv/Downloads/tensorrt-demo for image/tensorrt_demos-master (1)/tensorrt_demos-master/utils/yolov3.py", line 478, in detect
    trt_outputs, shape_orig_WH, conf_th)
  File "/home/cv/Downloads/tensorrt-demo for image/tensorrt_demos-master (1)/tensorrt_demos-master/utils/yolov3.py", line 111, in process
    outputs_reshaped, resolution_raw, conf_th)
  File "/home/cv/Downloads/tensorrt-demo for image/tensorrt_demos-master (1)/tensorrt_demos-master/utils/yolov3.py", line 161, in _process_yolo_output
    box, category, confidence = self._process_feats(output, mask)
  File "/home/cv/Downloads/tensorrt-demo for image/tensorrt_demos-master (1)/tensorrt_demos-master/utils/yolov3.py", line 232, in _process_feats
    col = col.reshape(grid_h, grid_w, 1, 1).repeat(3, axis=-2)
ValueError: cannot reshape array of size 4 into shape (10,2,1,1)

[trt_yolov3.txt](https://github.com/jkjung-avt/tensorrt_demos/files/4498934/trt_yolov3.txt)
[yolov3.txt](https://github.com/jkjung-avt/tensorrt_demos/files/4498938/yolov3.txt)
[yolov3-tiny-288-cfg.txt](https://github.com/jkjung-avt/tensorrt_demos/files/4498947/yolov3-tiny-288-cfg.txt)
 Many Thanks.

jkjung-avt, 2020-04-20:
I think you need to replace:

`    trt_yolov3 = TrtYOLOv3(args.model, (320, 64))`

with:

`    trt_yolov3 = TrtYOLOv3(args.model, (64, 320))`

in trt_yolov3.py.

jeansely, 2020-04-20:
Hi,
After moving the dimensions, the following problem occurred:
_ValueError: cannot reshape array of size 100 into shape (2,10,1,1)._

because:
[320/32=10 , 10*10=100]

So I realized that  have to be the same width and height.  By changing (64*320) to (320 * 320) in config file , I re-train the system and the problem was solved.

But the question is, Does the inference speed decrease?

Many Thanks.

jkjung-avt, 2020-04-20:
Do you not see any inference speed gain after optimizing the model with TensorRT (comparing to inference with Darknet directly)??

jeansely, 2020-04-20:
Hi,
I mean change the dimensions because the actual dimensions of my object are 64 x 320 (in tensorrt).

Many Thanks.

jkjung-avt, 2020-04-21:
Sure, when you reduce input image dimension, the total number of floating point computations of the model should be reduced.  So I would expect it to run faster.

closed 2020-04-23 by jkjung-avt

maky-hnou, 2020-06-03:
> Sure, when you reduce input image dimension, the total number of floating point computations of the model should be reduced. So I would expect it to run faster.

But also the accuracy would decrease, isn't it?

jkjung-avt, 2020-06-03:
In general, yes.  But it also depends on the size (in terms of width and height in pixels) of the objects you're targeting to detect.