jkjung-avt/tensorrt_demos #100: sort the location of objects

jeansely, 2020-04-26:
Hi,
How to sort the position of objects after detection?
i use yolo for character recognition and have to sort character after detection.
in below code i saw to sort by clss:
def draw_bboxes(self, img, boxes, confs, clss):
        """Draw detected bounding boxes on the original image."""
        for bb, cf, cl in zip(boxes, confs, clss):

where and how can sort by position of objects (for example left to right for car plate recognition).
so that the zip(boxes, confs, clss] sort only based on "x_min" of the boxes.

Many Thanks.

jeansely, 2020-04-28:
The best place in the visualization section and below:

def draw_bboxes(self, img, boxes, confs, clss):
"""Draw detected bounding boxes on the original image."""
          n = len(clss)
          plate = np.array([[0, 0, 0]])
          for bb, cf, cl in zip(boxes, confs, clss):
                     cl = int(cl)
                     x_min, y_min, x_max, y_max = bb[0], bb[1], bb[2], bb[3]
                     color = self.colors[cl]
                     cv2.rectangle(img, (x_min, y_min), (x_max, y_max), color, 2)
                     txt_loc = (max(x_min+2, 0), max(y_min+2, 0))
                     cls_name = self.cls_dict.get(cl, 'CLS{}'.format(cl))
                     txt = '{} {:.2f}'.format(cls_name, cf)
                     img = draw_boxed_text(img, txt, txt_loc, color)
             **     plate = np.append(plate, [[x_min, cf, cls_name]], axis = 0)
**    plate = np.delete(plate, 0, axis=0)
**    plate = plate[plate[:,0].astype(int).argsort()]
**    plate = plate[:,2]
return img

jkjung-avt, 2020-04-29:
Sorting by topleft x-coordinate of the bounding boxes should be straightforward.  You could do something like the following:

```
    # bb[0] is the x_min coordinate of the bbox
    sorted_indices = np.argsort([bb[0] for bb in bboxes])
    for i in sorted_indices:
        bb = bboxes[i]
        x_min, y_min, x_max, y_max = bb[0], bb[1], bb[2], bb[3]
        cl = int(clss[i])
        ......
```

closed 2020-05-01 by jkjung-avt