jkjung-avt/tensorrt_demos #104: build_engine.py: error: the following arguments are required: model

gpqls, 2020-04-29:
I want to run 'build_engine.py'
I run like this : `python3 build_engine.py
`
Then this error occured : 
`usage: build_engine.py [-h]
                       {ssd_mobilenet_v2_face,ssd_mobilenet_v1_coco,ssd_mobilenet_v1_egohands,ssd_mobilenet_v2_coco,ssd_mobilenet_v2_egohands}
build_engine.py: error: the following arguments are required: model`

So I run like this : `python3 build_engine.py --model ssd_mobilenet_v2_face
`
Then this error occured : 
`usage: build_engine.py [-h]
                       {ssd_mobilenet_v2_face,ssd_mobilenet_v1_coco,ssd_mobilenet_v1_egohands,ssd_mobilenet_v2_coco,ssd_mobilenet_v2_egohands}
build_engine.py: error: unrecognized arguments: --model
`
How to run 'build_engine.py' correctly?

jkjung-avt, 2020-04-29:
You need to define your own "ssd_mobilenet_v2_face" model in [MODEL_SPECS](https://github.com/jkjung-avt/tensorrt_demos/blob/master/ssd/build_engine.py#L22).

You could copy "ssd_mobilenet_v2_egohands" and then make necessary modifications.

gpqls, 2020-04-29:
I define my model : 

> 'ssd_mobilenet_v2_face': {
        'input_pb':   os.path.abspath(os.path.join(
                          DIR_NAME, 'ssd_mobilenet_v2_face.pb')),
        'tmp_uff':    os.path.abspath(os.path.join(
                          DIR_NAME, 'tmp_v2_face.uff')),
        'output_bin': os.path.abspath(os.path.join(
                          DIR_NAME, 'TRT_ssd_mobilenet_v2_face.bin')),
        'num_classes': 2,
        'min_size': 0.20000000298,
        'max_size': 0.949999988079,
        'input_order': [0, 2, 1],  # order of loc_data, conf_data, priorbox_data
    },

I have mad all the necessary modifications other than this.
Then I run like this : `python3 build_engine.py ssd_mobilenet_v2_face
`
Then this error occured : 

> [libprotobuf FATAL /externals/protobuf/aarch64/10.0/include/google/protobuf/repeated_field.h:1408] CHECK failed: (index) < (current_size_): 
Traceback (most recent call last):
  File "build_engine.py", line 235, in <module>
    main()
  File "build_engine.py", line 226, in main
    parser.parse(spec['tmp_uff'], network)
RuntimeError: CHECK failed: (index) < (current_size_): 

How to fix it?

jkjung-avt, 2020-04-29:
Does this happen to the "GridAnchor" layer? Similar to [this issue](https://forums.developer.nvidia.com/t/parsing-gridanchor-op-gridanchor-trt-protobuf-repeated-field-h-1408-check-failed-index-current-size/109683) on NVIDIA Developer forum?

```
[TensorRT] VERBOSE: UFFParser: Parsing GridAnchor[Op: GridAnchor_TRT].
[libprotobuf FATAL /externals/protobuf/x86_64/10.0/include/google/protobuf/repeated_field.h:1408] CHECK failed: (index) < (current_size):```

gpqls, 2020-04-29:
No, 
When i checked, there was another error like this : 
`[TensorRT] ERROR: Could not register plugin creator:  FlattenConcat_TRT in namespace: 
`
And this is total message.

> 2020-04-29 18:20:23.137856: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.0
[TensorRT] ERROR: Could not register plugin creator:  FlattenConcat_TRT in namespace: 
WARNING:tensorflow:From /usr/lib/python3.6/dist-packages/graphsurgeon/StaticGraph.py:125: The name tf.gfile.GFile is deprecated. Please use tf.io.gfile.GFile instead.

WARNING: To create TensorRT plugin nodes, please use the `create_plugin_node` function instead.
NOTE: UFF has been tested with TensorFlow 1.14.0.
WARNING: The version of TensorFlow installed on this system is not guaranteed to work with UFF.
UFF Version 0.6.5
=== Automatically deduced input nodes ===
[name: "Input"
op: "Placeholder"
attr {
  key: "dtype"
  value {
    type: DT_FLOAT
  }
}
attr {
  key: "shape"
  value {
    shape {
      dim {
        size: 1
      }
      dim {
        size: 3
      }
      dim {
        size: 300
      }
      dim {
        size: 300
      }
    }
  }
}
]
=========================================

Using output node NMS
Converting to UFF graph
Warning: No conversion function registered for layer: NMS_TRT yet.
Converting NMS as custom op: NMS_TRT
WARNING:tensorflow:From /usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:179: The name tf.AttrValue is deprecated. Please use tf.compat.v1.AttrValue instead.

Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_conf as custom op: FlattenConcat_TRT
Warning: No conversion function registered for layer: GridAnchor_TRT yet.
Converting MultipleGridAnchorGenerator as custom op: GridAnchor_TRT
Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_loc as custom op: FlattenConcat_TRT
DEBUG [/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:96] Marking ['NMS'] as outputs
No. nodes: 675
UFF Output written to /home/hyebin/tensorrt_demos/ssd/tmp_v2_face.uff
UFF Text Output written to /home/hyebin/tensorrt_demos/ssd/tmp_v2_face.pbtxt
[libprotobuf FATAL /externals/protobuf/aarch64/10.0/include/google/protobuf/repeated_field.h:1408] CHECK failed: (index) < (current_size_): 
Traceback (most recent call last):
  File "build_engine.py", line 235, in <module>
    main()
  File "build_engine.py", line 226, in main
    parser.parse(spec['tmp_uff'], network)
RuntimeError: CHECK failed: (index) < (current_size_): 

Jetpack 4.3 can install tensorflow only 1.15.0 and 2.1. I can't install tensorflow 1.14.0

gpqls, 2020-04-29:
Your code run very well. But when I run with my model, I got many error..

jkjung-avt, 2020-04-30:
If you want to use my code with minimum modifications, I would recommend to:

* Use **"6518c1c"** version of tensorflow object detection API
* Train and export the SSD model with tensorflow 1.12.x ~ 1.14.x

You could refer to https://github.com/jkjung-avt/tensorrt_demos/issues/76 for more details.

closed 2020-05-01 by jkjung-avt