jkjung-avt/tensorrt_demos #105: typeerror: a bytes-like object is required, not 'nonetype'

Darshcg, 2020-04-29:
I am trying Demo 4 on my Jetson Nano

I am facing error wrt Converting onnx to trt.
[image]

Here is the error

jkjung-avt, 2020-04-30:
I think the error `[TensorRT] ERROR: FAILED_ALLOCATION: std::bad_alloc` simply means TensorRT failed to allocate memory.

Please close your web browsers and other memory-hogging applications and try again.  If that still doesn't work, adding a swap might likely help.  Reference: [Setting up Jetson Nano: The Basics](https://jkjung-avt.github.io/setting-up-nano/)

Darshcg, 2020-04-30:
Thank you so much, Jung. I resolved it by closing all other applications. 
Btw After running yolov3-416 with TRT, I was getting 3 FPS, and yolov3-tiny-416 with TRT I was getting 16 FPS.
Btw how to run .weights and .cfg as it is(without TRT)? I want to test the Speed.

Thanks in Advance

jkjung-avt, 2020-04-30:
https://jkjung-avt.github.io/yolov3/

closed 2020-05-01 by jkjung-avt