jkjung-avt/tensorrt_demos #106: Don't support batch of image in the ssd_trt.py

PythonImageDeveloper, 2020-04-30:
Hi,
Thank you for sharing the codes.
I used jetpack 4.2.2 and opencv 3.4.6 and tensorflow 1.12.2.
and follow your codes for converting the model to uff and tensorrt model. but the structural of your code don't support batch of image processing at the same time. if predection time of one image takes 50 ms, the  predection time of batch of three image take 150 ms, why?
How do I solve the problem?

jkjung-avt, 2020-04-30:
I've built the TensorRT engine with [max_batch_size set to 1](https://github.com/jkjung-avt/tensorrt_demos/blob/master/ssd/build_engine.py#L204).  Please modify the code by yourself.

PythonImageDeveloper, 2020-04-30:
What's this :

>     def detect(self, img, conf_th=0.3):
>         """Detect objects in the input image."""
>         img_resized = _preprocess_trt(img, self.input_shape)
>         np.copyto(self.host_inputs[0], img_resized.ravel())
> 
>         cuda.memcpy_htod_async(
>             self.cuda_inputs[0], self.host_inputs[0], self.stream)
>         self.context.execute_async(
>             batch_size=1,
>             bindings=self.bindings,
>             stream_handle=self.stream.handle)
>         cuda.memcpy_dtoh_async(
>             self.host_outputs[1], self.cuda_outputs[1], self.stream)
>         cuda.memcpy_dtoh_async(
>             self.host_outputs[0], self.cuda_outputs[0], self.stream)
>         self.stream.synchronize()
> 
>         output = self.host_outputs[0]
>         return _postprocess_trt(img, output, conf_th, self.output_layout)

jkjung-avt, 2020-04-30:
Please refer to NVIDIA's original code here: https://github.com/AastaNV/TRT_object_detection/blob/master/main.py#L88

If you still have problems, please read TensorRT and PyCUDA documentation.

closed 2020-05-12 by jkjung-avt