jkjung-avt/tensorrt_demos #107: trt_outputs = [output.reshape(shape) for output, shape ValueError: cannot reshape array of size 92055 into shape (1,255,9,9)

Darshcg, 2020-04-30:
I am facing an error while running python3 trt_yolov3.py. I followed this https://github.com/jkjung-avt/tensorrt_demos/issues/51#issuecomment-580983383
but 92055/9/9= 1136.4814(Float value). 
Not Sure which value to put for "category_num" and "output_shapes" 

I am facing the error below
[image]

jkjung-avt, 2020-04-30:
What is the input shape of your model?  I guess it is not 288x288.

Darshcg, 2020-04-30:
I am using Yolov3 from Darkenet, not the Yolov3-416, 288).

jkjung-avt, 2020-04-30:
I use the number (608, 416 or 288) in the model name to determine input image shape.  Refer to this line of code: https://github.com/jkjung-avt/tensorrt_demos/blob/master/trt_yolov3.py#L81

Be sure to name the model correctly (or modify the python code based on your own model).

varungujarathi9, 2020-05-08:
I also got the same error with yolov3-416, but after investigating my model's input size I found out that it was 608 model and not 416, so try investigating the same. I got my model optimised with this repo.

closed 2020-05-12 by jkjung-avt