jkjung-avt/tensorrt_demos #108: ValueError: list.remove(x): x not in list

gpqls, 2020-05-04:
Traceback (most recent call last):
  File "build_engine.py", line 230, in <module>
    main()
  File "build_engine.py", line 207, in main
    spec)
  File "build_engine.py", line 183, in add_plugin
    graph.find_nodes_by_op("NMS_TRT")[0].input.remove("Input")
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/internal/containers.py", line 296, in remove
    self._values.remove(elem)
ValueError: list.remove(x): x not in list

While I run this repo, I got that error.
I saw that you suggest to use "6518c1c" version of tensorflow object detection API. 

I follow your code "hand-detection-tutorial". 
1. Tjen, am I using that version?
2.How to fix that error?

gpqls, 2020-05-04:
I solved this error
 remove the line : `self._values.remove(elem)
`

Then I got this error : 
[TensorRT] ERROR: UffParser: Validator error: TRTEngineOp_0: Unsupported operation _TRTEngineOp
[TensorRT] ERROR: Network must have at least one output
Traceback (most recent call last):
  File "build_engine.py", line 230, in <module>
    main()
  File "build_engine.py", line 224, in main
    buf = engine.serialize()
AttributeError: 'NoneType' object has no attribute 'serialize'

How to solve it?

gpqls, 2020-05-05:
I solve that error, but new error occured.

`python3 trt_ssd.py --model ssd_mobilenet_v2_face --image --filename ${HOME}/project/tf_trt_models/3.jpeg
`
Traceback (most recent call last):
  File "trt_ssd.py", line 102, in <module>
    main()
  File "trt_ssd.py", line 94, in main
    loop_and_detect(cam, trt_ssd, conf_th=0.3, vis=vis)
  File "trt_ssd.py", line 64, in loop_and_detect
    img = vis.draw_bboxes(img, boxes, confs, clss)
  File "/home/hyebin/project/tensorrt_demos/utils/visualization.py", line 96, in draw_bboxes
    color = self.colors[cl]
IndexError: list index out of range

What's the problem?

jkjung-avt, 2020-05-05:
You need to initialize "BBoxVisualization" with the correct "cls_dict" (class dictionary, which contains the correct number of classes).

https://github.com/jkjung-avt/tensorrt_demos/blob/master/trt_ssd.py#L92

gpqls, 2020-05-06:
Thank you for reply. But, I don't know how to modify code. My number of classes 1.

jkjung-avt, 2020-05-06:
On [line 86](https://github.com/jkjung-avt/tensorrt_demos/blob/master/trt_ssd.py#L86), replace this:

   ```
       cls_dict = get_cls_dict(args.model.split('_')[-1])
   ```

with this: (hard-coded class dictionary)

   ```
       cls_dict = {0: 'background', 1: 'boat'}
   ```

p.s. I had the impression that your custom model is detecting boats.  If not, just replace it with the correct class name.

closed 2020-05-12 by jkjung-avt

gpqls, 2020-05-12:
The error was resolved by that code you taught me.
But when I tried to run with USB webcam, there was an error.

> 2020-05-12 23:15:27.371447: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.0
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Internal data stream error.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Failed to open camera!

I used these commands
python3 --model ssd_mobilenet_v2_face --usb --vid 0

What's the problem?

jkjung-avt, 2020-05-12:
In case GStreamer v4l2src does not work for the USB webcam you are using, you could try to apply this workaround: https://github.com/jkjung-avt/tensorrt_demos/issues/8#issuecomment-538432571

gpqls, 2020-05-13:
I'm sorry to keep asking questions. 
I executed all the scripts perfectly with your help.
I want to have Jetson run the script automatically as it turns on.
I want to automatically run the trt_ssd.py file in the project/tensorrt_demos directory.
This is my command.
`python3 trt_ssd.py --model ssd_mobilenet_v2_face --usb --vid 0 --width 1280 --height 780`

I tried to start application, and I tried contrab.
But it didn't work out well because it  probably wrong.
Do you know how?

jkjung-avt, 2020-05-13:
Have you looked into using systemd and create a service to do that?  Reference: https://unix.stackexchange.com/a/398411/406831

Otherwise, I think you could find answers in [NVIDIA Developer Forum](https://forums.developer.nvidia.com/c/agx-autonomous-machines/55).