jkjung-avt/tensorrt_demos #110: 识别问题

312091526, 2020-05-13:
你好，我这边有两个问题：
1，我这边运行mtcnn demo的时候，发现如果是正常情况下，5个关键点的位置还是比较准确的，但是如果歪头的话，虽然人脸可以识别出来，但是5个关键点的位置就都标错了，感觉关键点的计算逻辑还是有问题的，能否帮忙解释一下？
2，除了输出人脸和5关键点，还有没有其他的输出信息？

jkjung-avt, 2020-05-13:
1. 關於"歪頭會造成5個關鍵點位置標錯"的問題，能否提供一兩張screenshot當作參考，以便確認問題?

2. MTCNN的出處在[這裡](https://kpzhang93.github.io/MTCNN_face_detection_alignment/)，原作者的設計是只有做bounding box和5個關鍵點的。

312091526, 2020-05-14:
1，提供了对应的截图
[image]

jkjung-avt, 2020-05-14:
我從Internet抓了一張圖來做測試，結果如下，看來是比較正常的。我想你的問題可能和你戴著口罩有關，若要解決的話，可能需要在MTCNN訓練資料中增加有「口罩」的圖片，並重新訓練model。

[image]

closed 2020-05-19 by jkjung-avt