jkjung-avt/tensorrt_demos #111: ModuleNotFoundError: No module named 'uff'

PatrikPatrik2, 2020-05-13:
I am a beginner, trying to run Demo #3: SSD on the nano.

I am using TF 1.15.2, and have been able to run object detection on models that I have trained at about 5 fps (mobilnet_v2).

When I run the ./build_engines.sh of Demo3
I get this error:

`+ for model in ssd_mobilenet_v1_coco ssd_mobilenet_v1_egohands ssd_mobilenet_v2_coco ssd_mobilenet_v2_egohands
+ python3 build_engine.py ssd_mobilenet_v1_coco
Traceback (most recent call last):
  File "build_engine.py", line 15, in <module>
    import uff
ModuleNotFoundError: No module named 'uff'
`
Is this because Demo3  is not compatible with TF1.15.2, or is it because I have not installed everything correctly?
BR
Patrik

PatrikPatrik2, 2020-05-13:
I use Jetpack 4.4

jkjung-avt, 2020-05-14:
I haven't tested JetPack-4.4.  But python "uff" module should be present on the system after you do a complete install of JetPack on your Jetson Nano.

Just for reference, in case you are downloading and installing TensorRT package (say, on a x86_64 PC), you will need to manually install "uff" as outlined in the [official documentation](https://docs.nvidia.com/deeplearning/sdk/tensorrt-archived/tensorrt_304/tensorrt-install-guide/index.html#installing-tar) (step 6): `$ cd TensorRT-x.x.x/uff; sudo pip3 install uff-0.2.0-py2.py3-none-any.whl`

PatrikPatrik2, 2020-05-14:
Thanks, could my problem be because I use a  virtual environment? Would I need to put something more in the file below?

`export PYTHONPATH=$PYTHONPATH:/home/pi/models/research:/home/pi/models/research/slim:/usr/lib/python3.6/dist-packages/cv2/python-3.6`

jkjung-avt, 2020-05-14:
Yes, that could be it.

Please try to copy "uff" module into your virtual environment.  For example, do the copying like this (but replace the virtual environment path with your own).

```
$ cp -r /usr/lib/python3.6/dist-packages/uff ${HOME}/venv/lib/python3.6/site-packages/
```

You'll likely have to do the same for "cv2" and maybe other modules.

PatrikPatrik2, 2020-05-14:
My understanding of the virtual enviroment is unfortunately very limited. For cv2 there was a file called cv2.so and by puting this :/usr/lib/python3.6/dist-packages/cv2/python-3.6 in the PYTHONPATH of the virtual enviroment I got CV2 to work. I did not find a similar file for UFF.

I do not find a special folder linked to my virtual enviroment - should I try to install uff inside my virtual enviroment? How do I do that?

jkjung-avt, 2020-05-14:
Do the copying as I've shown in https://github.com/jkjung-avt/tensorrt_demos/issues/111#issuecomment-628396960 should work.

PatrikPatrik2, 2020-05-14:
I do not have  the ${HOME}/venv directory, instead I used this cp -r /usr/lib/python3.6/dist-packages/uff /home/pi/.virtualenvs/py3cv4/lib/python3.6/site-packages/

this got me to a new error:
`(py3cv4) pi@jetson:~/project/tensorrt_demos/ssd$ ./build_engines.sh
+ for model in ssd_mobilenet_v1_coco ssd_mobilenet_v1_egohands ssd_mobilenet_v2_coco ssd_mobilenet_v2_egohands
+ python3 build_engine.py ssd_mobilenet_v1_coco
Traceback (most recent call last):
  File "build_engine.py", line 15, in <module>
    import uff
  File "/home/pi/.virtualenvs/py3cv4/lib/python3.6/site-packages/uff/__init__.py", line 2, in <module>
    from uff.converters.tensorflow.conversion_helpers import from_tensorflow  # noqa
  File "/home/pi/.virtualenvs/py3cv4/lib/python3.6/site-packages/uff/converters/tensorflow/conversion_helpers.py", line 12, in <module>
    from .converter_functions import *  # noqa
  File "/home/pi/.virtualenvs/py3cv4/lib/python3.6/site-packages/uff/converters/tensorflow/converter_functions.py", line 12, in <module>
    from uff.converters.tensorflow.converter import TensorFlowToUFFConverter as tf2uff
  File "/home/pi/.virtualenvs/py3cv4/lib/python3.6/site-packages/uff/converters/tensorflow/converter.py", line 13, in <module>
    from tensorflow.compat.v1 import AttrValue
ModuleNotFoundError: No module named 'tensorflow.compat'
`
When I run the ./install.sh I noticed that I also got an error
`ERROR: nvcc not found
`
What should I do?

jkjung-avt, 2020-05-14:
"ERROR: nvcc not found" -> You need to [add /usr/local/bin to your PATH](https://github.com/jkjung-avt/jetson_nano/blob/master/install_basics.sh#L8) environment variable.  You could refer to my blog post for more information: [Setting up Jetson Nano: The Basics](https://jkjung-avt.github.io/setting-up-nano/)

PatrikPatrik2, 2020-05-14:
That did not remove the problem
.
I never got tensorflow to work when I run your set-up, so I ended up of mergering your set-up with this https://www.pyimagesearch.com/2020/03/25/how-to-configure-your-nvidia-jetson-nano-for-computer-vision-and-deep-learning/ 
then I manage to have tensorflow up and running, but then your demo did not work.

I will start from scratch with your instructions again

PatrikPatrik2, 2020-05-15:
I installed according to https://jkjung-avt.github.io/jetpack-4.3/ and used TF 1.15.2 (with --user flag).
I got the demo to work.
But when I try to run "python build_engine.py"
I get this error:
`Traceback (most recent call last):
  File "build_engine.py", line 15, in <module>
    import uff
  File "/usr/lib/python2.7/dist-packages/uff/__init__.py", line 1, in <module>
    from uff import converters, model  # noqa
  File "/usr/lib/python2.7/dist-packages/uff/model/__init__.py", line 1, in <module>
    from . import uff_pb2 as uff_pb  # noqa
  File "/usr/lib/python2.7/dist-packages/uff/model/uff_pb2.py", line 6, in <module>
    from google.protobuf.internal import enum_type_wrapper
ImportError: No module named google.protobuf.internal
`
How can I correct this, or is this because I use jetpack 4.4 and TF 1.15.2?

jkjung-avt, 2020-05-15:
What do you mean by "I got the demo to work"?

And for the "google.protobuf" error, did you install protobuf-3.8.0 using my script?

I also tested JetPack-4.4 DP yesterday and today.  The results were OK.  You could refer to my exact steps in [JetPack-4.4 for Jetson Nano](https://jkjung-avt.github.io/jetpack-4.4/).

PatrikPatrik2, 2020-05-15:
Very sorry I am an idiot - I used python python build_engine.py; when I used python3 python build_engine.py I got the script to run

I have traind a SSD Mobilnet v2 model that works in normal tensorflow in jetson at about 5 fps.

I inserted:
`    'ssd_mobilenet_v2_PipeDot': {
        'input_pb':   os.path.abspath(os.path.join(
                          DIR_NAME, 'ssd_mobilenet_v2_PipeDot.pb')),
        'tmp_uff':    os.path.abspath(os.path.join(
                          DIR_NAME, 'tmp_ssd_mobilenet_v2_PipeDot.uff')),
        'output_bin': os.path.abspath(os.path.join(
                          DIR_NAME, 'TRT_ssd_mobilenet_v2_PipeDot.bin')),
        'num_classes': 2,
        'min_size': 0.05,
        'max_size': 0.95,
        'input_order': [0, 2, 1],  # order of loc_data, conf_data, priorbox_data
    },`

in build_engine.py

but when I run build_engine.py I got this error

`UFF Output written to /home/pi/project/tensorrt_demos/ssd/tmp_ssd_mobilenet_v2_PipeDot.uff
UFF Text Output written to /home/pi/project/tensorrt_demos/ssd/tmp_ssd_mobilenet_v2_PipeDot.pbtxt
[TensorRT] ERROR: UffParser: Validator error: Cast: Unsupported operation _Cast
[TensorRT] ERROR: Network must have at least one output
[TensorRT] ERROR: Network validation failed.
Traceback (most recent call last):
  File "build_engine.py", line 230, in <module>
    main()
  File "build_engine.py", line 224, in main
    buf = engine.serialize()
AttributeError: 'NoneType' object has no attribute 'serialize'
`

PatrikPatrik2, 2020-05-15:
What have I done wrong?

joelosw, 2020-05-15:
@PatrikPatrik2 Replace "ToFloat" with "Cast" in the namespace_plugin_map to get rid of the "Cast" error. 
This is due to a new tf version, that you are probably using.

PatrikPatrik2, 2020-05-15:
When I do the change, I do not get the error any more, but after quite some time it aborts and writes:
`[TensorRT] INFO: Some tactics do not have sufficient workspace memory to run. Increasing workspace size may increase performance, please check verbose output.
[TensorRT] INFO: Detected 1 inputs and 2 output network tensors.
#assertionnmsPlugin.cpp,249
Aborted (core dumped)`

A very strange thing is that when I use the demo model ssd_mobilenet_v2_egohands.pb I get the error:
`AttributeError: 'NoneType' object has no attribute 'serialize'`

but when I change back "ToFloat" in namespace_plugin_map it works again.

What shall I do?

PatrikPatrik2, 2020-05-16:
Have I understood it correctly that this will only work if I train my model in TF 1.12 before running build_engine.py?
If so this is very depressing for me, as I do now own a GPU and do my training in Colab, where I only get access to TF1.15.2

joelosw, 2020-05-16:
So, the first Error you got (the "Cast"-Error) happens when you are doing the first step, that is creating the uff file. The uff-converter does not recognize the "Cast" operation, hence you'll have to tell explicitly to plugin the Input node instead. 
If that is working, the assertion error happens at a later stage. So if the program already fails at an earlier stage and you will never get the assertion error. 

The assertion error indicates, that the number of generated priorboxes does not match the number of confidences that were generated by the network. 
I had the same error on one of my own networks. I fixed it by changing the depth_multiplier to 0.25 in the config-file of the model. Maybe this will work for you as well, but honestly I am not sure why this worked for me.

PatrikPatrik2, 2020-05-16:
Thanks, I have now started a new training where I changed depth_multiplier from 1 to 0.25

In Colab I manage to covert frozen_graph.pb to .uff by using:
`!python3 /usr/lib/python3.6/dist-packages/uff/bin/convert-to-uff.py frozen_inference_graph.pb -O NMS -p config.py`

I do not really know how to use this file, it seems that build_engine.py generates some additional files

PatrikPatrik2, 2020-05-16:
Did not work :(

`#assertionnmsPlugin.cpp,246
Aborted (core dumped)
`
depth_multiplier = 0.25
namespace_plugin_map changed to "Cast"

jkjung-avt, 2020-05-17:
@PatrikPatrik2 For the error of `[TensorRT] INFO: Some tactics do not have sufficient workspace memory to run. Increasing workspace size may increase performance, please check verbose output. [TensorRT] INFO: Detected 1 inputs and 2 output network tensors. #assertionnmsPlugin.cpp,249 Aborted (core dumped)`, could you turn on debug logging and try again?  (You need to also add [this Cast patch](https://github.com/jkjung-avt/tensorrt_demos/pull/114/commits/d408ac60d42f7f0b55598a3c005e6ace7c7989a4#diff-c3c3c939b36fc730d144382e171bc3c4R162) into the code, as you've done previously.)

You could enable TensorRT optimization debug logging by setting "DEBUG_UFF" to True:
https://github.com/jkjung-avt/tensorrt_demos/blob/master/ssd/build_engine.py#L73

PatrikPatrik2, 2020-05-17:
Below is what I got after changeing according to your request:
`DEBUG [/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:96] Marking ['NMS'] as outputs
No. nodes: 644
UFF Output written to /home/pi/project/tensorrt_demos/ssd/tmp_v2_Dot.uff
UFF Text Output written to /home/pi/project/tensorrt_demos/ssd/tmp_v2_Dot.pbtxt
[TensorRT] INFO: Some tactics do not have sufficient workspace memory to run. Increasing workspace size may increase performance, please check verbose output.
[TensorRT] INFO: Detected 1 inputs and 2 output network tensors.
#assertionnmsPlugin.cpp,249
Aborted (core dumped)
`

jkjung-avt, 2020-05-18:
Here's the source code which corresponds to the error of `#assertionnmsPlugin.cpp,249 Aborted`.

https://github.com/NVIDIA/TensorRT/blob/master/plugin/nmsPlugin/nmsPlugin.cpp#L249

```
    ASSERT(numPriors * param.numClasses == inputDims[param.inputOrder[1]].d[0]);
```

So you need to check and make sure you have set these 2 numbers correctly for your custom model: (a) [numClasses](https://github.com/jkjung-avt/tensorrt_demos/blob/master/ssd/build_engine.py#L66), (b) [inputOrder](https://github.com/jkjung-avt/tensorrt_demos/blob/master/ssd/build_engine.py#L69).

PatrikPatrik2, 2020-05-18:
In my custom model, in the config file I had:
`model {
  ssd {
    num_classes: 2`

so in build_engine.py I changed to 
`'num_classes': 2,`

The input order I do not know, so I tested all 6 possible combinations of 
`'input_order': [1, 0, 2],`
(1,0,2; 1,2,0 etc)

all 6 combinations of input order resulted in:

`[TensorRT] INFO: Detected 1 inputs and 2 output network tensors.
#assertionnmsPlugin.cpp,246
Aborted (core dumped)
pi@jetson:~/project/tensor`

Any thing else I can try?

jkjung-avt, 2020-05-18:
'num_classes' should be 3 in your case, since you need to add "background" as an additional class.

As to 'input_order', you could figure it out by looking at the pbtxt file.  I briefly described it in [my blog post](https://jkjung-avt.github.io/tensorrt-ssd/).  Otherwise, you could refer to [this discussion](https://forums.developer.nvidia.com/t/how-to-write-config-py-for-converting-ssd-mobilenetv2-to-uff-format/73376/8?u=jkjung13) on NVIDIA Developer Forum.

In case you are still having trouble with 'input_order', please post your pbtxt file and I'll check it out.

PatrikPatrik2, 2020-05-18:
From the pbtxt file I got 
`    fields {
      key: "inputOrder_u_ilist"
      value {
        i_list {
          val: 1
          val: 0
          val: 2
        }`

I updated the build_engine.py according to below and used the order above as input order, and renamed my frozen graph to ssd_mobilenet_v2_egohands.pb.

`    'ssd_mobilenet_v2_egohands': {
        'input_pb':   os.path.abspath(os.path.join(
                          DIR_NAME, 'ssd_mobilenet_v2_egohands.pb')),
        'tmp_uff':    os.path.abspath(os.path.join(
                          DIR_NAME, 'tmp_v2_egohands.uff')),
        'output_bin': os.path.abspath(os.path.join(
                          DIR_NAME, 'TRT_ssd_mobilenet_v2_egohands.bin')),
        'num_classes': 3,
        'min_size': 0.05,
        'max_size': 0.95,
        'input_order': [1, 0, 2],  # order of loc_data, conf_data, priorbox_data
    },`

result was the same :(

`[TensorRT] INFO: Some tactics do not have sufficient workspace memory to run. Increasing workspace size may increase performance, please check verbose output.
[TensorRT] INFO: Detected 1 inputs and 2 output network tensors.
#assertionnmsPlugin.cp`

jkjung-avt, 2020-05-18:
Please look for the following 3 lines in the NMS node in the pbtxt file (the pbtxt file is generated after you run build_engine.py).  The order of the 3 lines for your model might not match the example below.  Please post what you see.

```
{
   ...
   input: "concat_priorbox"
   input: "Squeeze"
   input: "concat_box_conf"
}
```

PatrikPatrik2, 2020-05-18:
` inputs: "Squeeze"
    inputs: "concat_priorbox"
    inputs: "concat_box_conf"
    operation: "_NMS_TRT"`

jkjung-avt, 2020-05-18:
You should set 'input_order' to '[0, 2, 1]'.

p.s. [loc_data, conf_data, priorbox_data]

PatrikPatrik2, 2020-05-19:
The .bin file was sucessfully saved!

However, when I run trt_ssd.py I got this, and thereafter it froze

`[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
Traceback (most recent call last):
  File "trt_ssd.py", line 101, in <module>
    main()
  File "trt_ssd.py", line 93, in main
    loop_and_detect(cam, trt_ssd, conf_th=0.3, vis=vis)
  File "trt_ssd.py", line 63, in loop_and_detect
    img = vis.draw_bboxes(img, boxes, confs, clss)
  File "/home/pi/project/tensorrt_demos/utils/visualization.py", line 96, in draw_bboxes
    color = self.colors[cl]
IndexError: list index out of range`

jkjung-avt, 2020-05-19:
You need to define your own object class list/dictionary.  For example, replace [this line](https://github.com/jkjung-avt/tensorrt_demos/blob/master/trt_ssd.py#L86):

```
    cls_dict = get_cls_dict(args.model.split('_')[-1])
```

with:

```
    cls_dict = {0: 'background', 1: 'class1', 2: 'class2'}
```

PatrikPatrik2, 2020-05-19:
31 FPS running the async verion!!!
Thank you ever so much, extremely greatfull for your support!!!!

By the way, is it possible  to increase the frame rate of the camera by changing your code? I use a pi cam, and higher frame rate seems to make a clearer picture (like using a shorter shutter time)

I tested 120 fps
`gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM),width=640, height=480, framerate=120/1, format=NV12' ! nvvidconv flip-method=0 ! 'video/x-raw,width=640, height=480' ! nvvidconv ! nvegltransform ! nveglglessink -e
`

jkjung-avt, 2020-05-19:
Modify the code [here](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/camera.py#L114).

PatrikPatrik2, 2020-05-19:
Perfect, thanks!!!

My next task is to get the coordinates of each boxes in each frame and save them to an csv file (one frame per line, the complete session in one csv file) such that I can analyse them in excel.
I thinking something like:
`millis or frame number 1; Classification_1; Probability_1; xCoordinate1_1; xCoordinate2_1; yCoordinate1_1; yCoordinate2_1; Classification_2; Probability_2; xCoordinate1_2; xCoordinate2_2; yCoordinate1_2; yCoordinate2_2; Classification_1; Probability_1; xCoordinate1_1; xCoordinate2_1; yCoordinate1_1; yCoordinate2_1; Classification_3; Probability_3; xCoordinate1_3; xCoordinate2_3; yCoordinate1_3; yCoordinate2_3

millis or frame number 2; Classification_1; Probability_1; xCoordinate1_1; xCoordinate2_1; yCoordinate1_1; yCoordinate2_1;`

I would be extremely grateful if you could give me some guidance how I can incorporate this into your code

jkjung-avt, 2020-05-19:
This is not exactly what you're asking for, but should be helpful: http://disq.us/p/298nabk

Otherwise, you'll need to read and modify the code by yourself.

PatrikPatrik2, 2020-05-26:
Very grateful, that lead me in the right direction to solve my problem - Thanks

Now I have only one more question, I like to make a "digital zoom" of the image (i use an onboard pi cam) - cropping away a large part of the image. Using CV2 i assume that I can use soemthing like:
`img=img.crop((300, 400, 600, 700)`

I tried to insert this into camera.py, but I have not been sucessful. Do you know where I shall put the code to crop the image (before object detection and displaying the picture)?

jkjung-avt, 2020-05-26:
https://opencv-python-tutroals.readthedocs.io/en/latest/py_tutorials/py_core/py_basic_ops/py_basic_ops.html#image-roi

closed 2020-05-27 by jkjung-avt

TUT-jiayou, 2021-07-23:
Hi,all

How to install a python deb package to a anaconda virtual environment? Not move it manual.

sachinkmohan, 2022-03-08:
The way I solved it on my PC.
 - Go to the TensorRT [download page](https://developer.nvidia.com/nvidia-tensorrt-download) and download the tar file according to your version.
 - Extract it. There you will find a uff folder. `cd uff`
 - I was using the virtual environment. So I directily executed `pip install uffxxxx.whl` file.
 - That's it! It should work.