jkjung-avt/tensorrt_demos #112: Segmentation fault when running tutorial

WillJStone, 2020-05-14:
Hi there,

First of all, thanks so much for making these demos. I've been struggling with getting tensorrt to work for a while now, and this is the closest I have gotten.

I am using Jetpack 4.2.2 on a jetson nano and I am trying to run your tutorial without any modifications. I am able to run the build_engines.sh script seemingly without error, but then when I go to run the trt_ssd.py demo, I am running into a segmentation fault. Here is all of the output:

GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
Segmentation fault (core dumped)

Any advice anyone may have regarding this would be greatly appreciated.

jkjung-avt, 2020-05-14:
I think this error message, "GST_ARGUS: Available Sensor modes : Segmentation fault (core dumped)", means that you are trying to use the onboard (CSI) camera but you don't have one installed on your Jetson Nano.

Please test with an image file or a video file instead.  Examples:

```
$ cd ${HOME}/project/tensorrt_demos
$ python3 trt_ssd.py --model ssd_mobilenet_v1_coco --image --filename huskies.jpg
$ python3 trt_ssd.py --model ssd_mobilenet_v1_coco --file --filename input.mp4
```

WillJStone, 2020-05-14:
I guess that's what I get for not taking a closer look at the source! 

Thanks so much my friend.

closed 2020-05-14 by WillJStone