jkjung-avt/tensorrt_demos #113: run build_engine.py on ssdlite_mobilenet_v3_kitti error

ynjiun, 2020-05-14:
Hi,

It's impressive about your demo on ssd. I am able to run your demo ssd examples without problem.

Then I tried to add one model "ssdlite_mobilenet_v3_kitti" in build_engine.py and encounter following error. Attached please find info.zip (which includes my ssdlite_mobilenet_v3_kitti.pb, revised build_engine.py and errmsg) for your information for duplication. Could you please check what's wrong and how to fix it? Thanks a lot.

[info.zip](https://github.com/jkjung-avt/tensorrt_demos/files/4625963/info.zip)

Error message:
paul@nano:~/jetson-inference/demo/tensorrt_demos/ssd$ python3 build_engine.py ssdlite_mobilenet_v3_kitti
2020-05-13 20:32:40.207888: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.0
[TensorRT] ERROR: Could not register plugin creator:  FlattenConcat_TRT in namespace: 
WARNING:tensorflow:From /usr/lib/python3.6/dist-packages/graphsurgeon/StaticGraph.py:125: The name tf.gfile.GFile is deprecated. Please use tf.io.gfile.GFile instead.

WARNING: To create TensorRT plugin nodes, please use the `create_plugin_node` function instead.
NOTE: UFF has been tested with TensorFlow 1.14.0.
WARNING: The version of TensorFlow installed on this system is not guaranteed to work with UFF.
UFF Version 0.6.5
```
Automatically deduced input nodes 
[name: "Input"
op: "Placeholder"
input: "Cast"
attr {
  key: "dtype"
  value {
    type: DT_FLOAT
  }
}
attr {
  key: "shape"
  value {
    shape {
      dim {
        size: 1
      }
      dim {
        size: 3
      }
      dim {
        size: 320
      }
      dim {
        size: 320
      }
    }
  }
}
]
```

Using output node NMS
Converting to UFF graph
Warning: No conversion function registered for layer: NMS_TRT yet.
Converting NMS as custom op: NMS_TRT
WARNING:tensorflow:From /usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:179: The name tf.AttrValue is deprecated. Please use tf.compat.v1.AttrValue instead.

Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_loc as custom op: FlattenConcat_TRT
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV3/layer_13_2_Conv2d_5_3x3_s2_128/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV3/layer_13_2_Conv2d_5_3x3_s2_128_depthwise/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV3/layer_13_1_Conv2d_5_1x1_64/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV3/layer_13_2_Conv2d_4_3x3_s2_256/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV3/layer_13_2_Conv2d_4_3x3_s2_256_depthwise/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
[... 173 lines cut ...]
Converting concat_box_conf as custom op: FlattenConcat_TRT
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting BoxPredictor_5/BoxEncodingPredictor_depthwise/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting BoxPredictor_4/BoxEncodingPredictor_depthwise/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting BoxPredictor_3/BoxEncodingPredictor_depthwise/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting BoxPredictor_2/BoxEncodingPredictor_depthwise/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting BoxPredictor_1/BoxEncodingPredictor_depthwise/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting BoxPredictor_0/BoxEncodingPredictor_depthwise/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
DEBUG [/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:96] Marking ['NMS'] as outputs
No. nodes: 864
UFF Output written to /home/paul/jetson-inference/demo/tensorrt_demos/ssd/tmp_v3_kitti.uff
UFF Text Output written to /home/paul/jetson-inference/demo/tensorrt_demos/ssd/tmp_v3_kitti.pbtxt
[TensorRT] ERROR: UffParser: Validator error: Cast: Unsupported operation _Cast
[TensorRT] ERROR: Network must have at least one output
Traceback (most recent call last):
  File "build_engine.py", line 239, in <module>
    main()
  File "build_engine.py", line 233, in main
    buf = engine.serialize()
AttributeError: 'NoneType' object has no attribute 'serialize'

jkjung-avt, 2020-05-14:
I don't have time to check your model.

But for the problem of "FusedBatchNormV3", please refer to https://github.com/jkjung-avt/tensorrt_demos/issues/76#issuecomment-607879831

And for the problem of "Unsupported operation _Cast", you could refer to https://github.com/jkjung-avt/tensorrt_demos/issues/33#issuecomment-570782526

joelosw, 2020-05-15:
@ynjiun Replace "ToFloat" with "Cast" in the namespace_plugin_map. 
It was renamed in tf 1.15, I guess you trained your model with 1.15, so by replacing the name it should be fixed.

ynjiun, 2020-05-16:
@joelosw yes I use tf 1.15. could you please give me a code sample on how to replace "ToFloat" with "Cast"? Thanks a lot for your help.

joelosw, 2020-05-17:
I think it will only fix one of your errors. 
But in the build_enginy.py script, there is a dictionary "namespace_plugin_map". 
When you add a line to replace the "Cast" namespace that should at least fix the "Cast" error in your code.

`namespace_plugin_map = {
        "MultipleGridAnchorGenerator": PriorBox,
        "Postprocessor": NMS,
        "Preprocessor": Input,
        "ToFloat": Input,
        "Cast": Input, #added for models trained with tf 1.15
        "image_tensor": Input,
        "MultipleGridAnchorGenerator/Concatenate": concat_priorbox,  # for 'ssd_mobilenet_v1_coco'
        "Concatenate": concat_priorbox,  # for other models
        "concat": concat_box_loc,
        "concat_1": concat_box_conf
    }`

ynjiun, 2020-05-17:
Thank you @joelosw 
Yes, your code successfully remove the 'Cast' error.
Next error show up is 
`[TensorRT] ERROR: UffParser: Validator error: BoxPredictor_2/ClassPredictor_depthwise/BatchNorm/FusedBatchNormV3: Unsupported operation _FusedBatchNormV3`
Found code from web to fix this as well:
```
def replace_fusedbnv3(graph):
    """Replace all 'FusedBatchNormV3' in the graph with 'FusedBatchNorm'.
    NOTE: 'FusedBatchNormV3' is not supported by UFF parser.
    https://devtalk.nvidia.com/default/topic/1066445/tensorrt/tensorrt-6-0-1-tensorflow-1-14-no-conversion-function-registered-for-layer-fusedbatchnormv3-yet/post/5403567/#5403567
    """
    for node in graph.find_nodes_by_op('FusedBatchNormV3'):
        gs.update_node(node, op='FusedBatchNorm')
    return graph
```
and call it in add_plugin:
```
  if model == 'ssdlite_mobilenet_v3_kitti':
        #graph.find_nodes_by_name("Input")[0].input.remove("Cast")
        graph = replace_fusedbnv3(graph)
```
And not hopefully is the final error:
`[TensorRT] ERROR: UffParser: Validator error: FeatureExtractor/MobilenetV3/expanded_conv_7/depthwise/hard_swish/add: Unsupported operation _AddV2`
Any suggestion on this one?
Thanks again for your help!

jkjung-avt, 2020-05-18:
Similar to replacing "FusedBatchNormV3" with "FusedBatchNorm", I think you could also replace all "ADDV2" with "ADD" using graphsurgeon.

ynjiun, 2020-05-18:
Thank you @jkjung-avt . Yes get AddV2 resolved. Now the status is:
DEBUG [/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:96] Marking ['NMS'] as outputs
No. nodes: 863
UFF Output written to /home/paul/jetson-inference/demo/tensorrt_demos/ssd/tmp_v3_kitti.uff
UFF Text Output written to /home/paul/jetson-inference/demo/tensorrt_demos/ssd/tmp_v3_kitti.pbtxt
[TensorRT] WARNING: Calling isShapeTensor before the entire network is constructed may result in an inaccurate result.
[TensorRT] ERROR: FeatureExtractor/MobilenetV3/expanded_conv/squeeze_excite/mul: elementwise inputs must have same dimensions or follow broadcast rules (input dimensions were [80,80,16] and [16,1,1])
python3: ../builder/cudnnBuilderUtils.cpp:577: void nvinfer1::cudnn::updatePaddingAccordingToPaddingMode(nvinfer1::PaddingMode, nvinfer1::Dims&, nvinfer1::Dims&, nvinfer1::Dims, nvinfer1::Dims, nvinfer1::Dims, nvinfer1::Dims, bool): Assertion `inputDims.nbDims >= nbSpatialDims' failed.

Is this because this is ssdlite which uses the depthwise convolution that tensorRT doesn't support? or other trivial reason which we can fix easily? Any suggestion? Thanks a lot.

jkjung-avt, 2020-05-18:
The error message is pretty clear.  You need to figure out why and fix the model (possibly with graphsurgeon).

```
[TensorRT] ERROR: FeatureExtractor/MobilenetV3/expanded_conv/squeeze_excite/mul: elementwise inputs must have same dimensions or follow broadcast rules (input dimensions were [80,80,16] and [16,1,1])
```

ynjiun, 2020-05-19:
well, this frozen model (ssdlite_mobilenet_v3_kitti.pb) was tested and run well under tensorflow-gpu==1.15.0 and numpy==1.16.0. That implies the mul elementwise should be correct (at least at numpy==1.16.0 with tf1.15)  No sure whatelse need to be fixed except for naming and ops compatibility for tensorRT. Perhaps tensorRT is not ready for ssdlite_mobilenet_v3 yet ; (

jkjung-avt, 2020-05-19:
I wondered whether the [16, 1, 1] tensor should have been reshaped to [1, 1, 16] (in which case it could be broadcast to [80, 80, 16]), while the reshape operation somehow got dropped during the conversion (PB -> UFF & graphsurgeon).

If that's the case, you could try to add a reshape node in the UFF (using graphsurgeon again) to make it work.

PythonImageDeveloper, 2020-05-28:
Hi all, @jkjung-avt @joelosw 
You prefer to replace "ToFloat" with "Cast", I have some question:
How to you know we must change  "ToFloat" with "Cast"? because "ToFloat" op is replaced in the TF1.15? and the preptrained model before trained with older version of TF that has 'Cast' op? If so, when we use a pretrained model and we want to train this model with new version of TF, the TF replace all op with new names?

joelosw, 2020-05-28:
Hi @PythonImageDeveloper, 
you are right, the "ToFloat" was replaced with "Cast" in tf 1.15. 
So the older frozen graph have "ToFloat" namespaces, while the new graphs have a "Cast" namespace. 

So in general it  is not necessary to replace all operations, you just need to add the ones that were renamed in tensorflow. 

2. It is the other way around. So the Key determines for which namespace to look for in the tf graph. The operations of this namespace are then replaced by the uff modules (NMS, ...) to convert it to the uff format. The uff operations were defined before defining the dictionary.

PythonImageDeveloper, 2020-05-28:
Thanks,
If I wrong understand please correct me.
for four "Input" in the dictionary,  "ToFloat" key in used for old tf graph and "Cast" key is used for new tf graph, right? and "Preprocessor" and  "image_tensor" also are for other version of tf graph? if not so, so the "Input" is new node that  will define in UFF graph, why we assign two different node in tf garph like  "Preprocessor" and  "image_tensor" to one node in UFF like "Input"?

jkjung-avt, 2020-05-29:
> 2- in the dictionary of below, the keys are related to UFF ops and values are ops of original graph? and this way change the the name of ops from value to key? if so, why you use many keys for input value?
> 
>     namespace_plugin_map = {
>         "MultipleGridAnchorGenerator": PriorBox,
>         "Postprocessor": NMS,
>         "Preprocessor": Input,
>         "ToFloat": Input,
>         "Cast": Input, #added for models trained with tf 1.15
>         "image_tensor": Input,
>         "MultipleGridAnchorGenerator/Concatenate": concat_priorbox,  # for 'ssd_mobilenet_v1_coco'
>         "Concatenate": concat_priorbox,  # for other models
>         "concat": concat_box_loc,
>         "concat_1": concat_box_conf
>     }

Please refer to official documentation of [collapse_namespaces()](https://docs.nvidia.com/deeplearning/tensorrt/api/python_api/graphsurgeon/graphsurgeon.html#dynamic-graph-inherits-from-staticgraph).

The "keys" in this dictionary are nodes in the original tensorflow graph, while "values" ops which we want in the final UFF graph.  After we do collapse_namespace() with this:

```
    namespace_plugin_map = {
        ......
        "ToFloat": Input,
        "Cast": Input, #added for models trained with tf 1.15
        ......
    }
```

All nodes named "ToFloat", "ToFloat/*", "Cast" or "Cast/*" in the original tensorflow graph would be removed and merged into the "Input" plugin node which [we created in the code earlier](https://github.com/jkjung-avt/tensorrt_demos/blob/master/ssd/build_engine.py#L95).

closed 2020-08-11 by jkjung-avt

sabaridsn, 2021-11-05:
@ynjiun are you able to convert SSD mobilenet v3 lite into  TensorRT engine