jkjung-avt/tensorrt_demos #116: bin and etlt file

PythonImageDeveloper, 2020-05-17:
Hi @jkjung-avt 
Is it possible to use .bin and .etlt files of TLT to run in TensorRT?

jkjung-avt, 2020-05-17:
TLT should be using TensorRT as the underlying technology.  But its saved file format does not seem to be a direct serialization of the TensorRT engine.  I don't have much experience working with that.

PythonImageDeveloper, 2020-05-17:
https://devblogs.nvidia.com/training-custom-pretrained-models-using-tlt/

> his post walks you through the workflow, from downloading the TLT Docker container and AI models from NVIDIA NGC, to training and validating with your own dataset and then exporting the trained model for deployment on the edge using NVIDIA DeepStream SDK and NVIDIA TensorRT. Alternatively, these models can be exported and converted to a TensorRT engine for deployment. We describe each of the models, four detection and two classification models.

closed 2020-05-19 by jkjung-avt