jkjung-avt/tensorrt_demos #117: Custom frozen graph gives the following error

saurabbhsp, 2020-05-17:
Traceback (most recent call last):
  File "build_engine.py", line 218, in <module>
    main()
  File "build_engine.py", line 195, in main
    spec)
  File "build_engine.py", line 173, in add_plugin
    graph.find_nodes_by_name("Input")[0].input.remove("image_tensor:0")
ValueError: remove(x): x not in container

For generation of frozen graphs, I am using export_Inference_graph.py

jkjung-avt, 2020-05-18:
Try commenting out this line from the code.

`graph.find_nodes_by_name("Input")[0].input.remove("image_tensor:0")`

jkjung-avt, 2020-05-27:
Has the issue been resolved?

saurabbhsp, 2020-05-27:
Yes it is resolved. Thanks for your help. The issue with custom training is that you need to train the model with TF-1.10, using newer versions results in the above mentioned error.

Thanks for your help.

closed 2020-05-27 by saurabbhsp