jkjung-avt/tensorrt_demos #118: cv2.resize in frozen graph

PythonImageDeveloper, 2020-05-18:
Hi @jkjung-avt 
As you know, resize operation in the resource constraint devices has expensive cost.
I don'w know why use resize operation before feeding image to graph. I feed the image without resize operation and the results is correctly and take a same time for prediction in all resolution. I think the graph has resizer node or operation, what's your opinion?

jkjung-avt, 2020-05-18:
Which model are you talking about??

PythonImageDeveloper, 2020-05-19:
SSDs

jkjung-avt, 2020-05-19:
I think the tensorflow SSD do have "image_resizer" built-in in the models.  See [ssd_mobilenet_v1_coco.config](https://github.com/tensorflow/models/blob/master/research/object_detection/samples/configs/ssd_mobilenet_v1_coco.config#L43) as an example.

I myself prefer using "cv2" to resize input images since I recall that runs a little bit faster.

This is kind of a design choice.  I'm sure cv2.resize() runs on the CPU, but I'm not sure about tensorflow "image_resizer".  By using cv2.resize() we could let GPU focus on NN operations, and potentially parallelized image preprocessing with CPU.  That's exactly what I did for the "async" code.

PythonImageDeveloper, 2020-05-19:
Thanks.
Is it possible to remove the image_resizer node from Network when we sant to create tensorRT model?

jkjung-avt, 2020-05-20:
Sure.  You should be able to remove unwanted nodes in the graph (before converting it to UFF) by calling [graphsurgeion's](https://docs.nvidia.com/deeplearning/tensorrt/api/python_api/graphsurgeon/graphsurgeon.html) remove(), forward_inputs() or collapse_namespaces() API.

closed 2020-05-27 by jkjung-avt