jkjung-avt/tensorrt_demos #119: ModuleNotFoundError: No module named 'pytrt'

mariusmotea, 2020-05-23:
Hi,

i want to test the face detection speed of your implementation but it seams this module is missing. I have same board, same JetPack version.

```
(py3cv4) marius@Jetson:~/Documents/test/tensorrt_demos$ python3 trt_mtcnn.py --image --filename ${HOME}/Pictures/avengers.jpg
Traceback (most recent call last):
  File "trt_mtcnn.py", line 14, in <module>
    from utils.mtcnn import TrtMtcnn
  File "/home/marius/Documents/test/tensorrt_demos/utils/mtcnn.py", line 6, in <module>
    import pytrt
ModuleNotFoundError: No module named 'pytrt'
```

jkjung-avt, 2020-05-23:
The "pytrt" module should be built after you do `make` in the tensorrt_demos/ directory.

Thanks for pointing this out.  I should add that into the README.

closed 2020-05-27 by jkjung-avt

yumingchen, 2021-03-26:
cd path/tensorrt_demos/
make

tm-topnet, 2022-07-21:
why it is avalable import pytrt only from that dir ?
i'm working with jetson xavier... and import pytrt works only from that dir.
how to make import from another directory?

jkjung-avt, 2022-07-21:
After you build the cython code with `make`, there will be a "pytrt.cpython-3?m-aarch64-linux-gnu.so" file generated.

Just copy that file to the directory where you'd like to `import pytrt` from.

EdithZou, 2024-11-12:
Hi, I am trying to use x86_64 architecture instead of ARM64 , and I wonder is there any solutions of generating a pytrt module file for x86_64 system? Thanks!

jkjung-avt, 2024-11-12:
> Hi, I am trying to use x86_64 architecture instead of ARM64 , and I wonder is there any solutions of generating a pytrt module file for x86_64 system? Thanks!

Please refer to [README_x86.md](https://github.com/jkjung-avt/tensorrt_demos/blob/master/README_x86.md).  You could build the pytrt module from source code in this repo.

jkjung-avt, 2024-11-19:
Did you see any error when building the TensorRT MTCNN engines?

```
$ cd ${HOME}/project/tensorrt_demos/mtcnn
$ make
$ ./create_engines
```

Also, did you see any error when building the Cython code?

```
$ sudo pip3 install Cython
$ cd ${HOME}/project/tensorrt_demos
$ make
```

EdithZou, 2024-11-19:
> Did you see any error when building the TensorRT MTCNN engines?
> 
> ```
> $ cd ${HOME}/project/tensorrt_demos/mtcnn
> $ make
> $ ./create_engines
> ```
> 
> Also, did you see any error when building the Cython code?
> 
> ```
> $ sudo pip3 install Cython
> $ cd ${HOME}/project/tensorrt_demos
> $ make
> ```

The above errors occurred when I tried to run ./create_engines after make. And when I use TensorRT 7.1.3.4, no error occurs when running make in /tensorrt_demos folder. When using other versions, for example TensorRT 10.x, some .h files are deprecated, so errors occur. But the problem is that TensorRT 7.1.3.4 only supports CUDA 10.2 and below. However my GPU (Geforce RTX 4060) supports at least CUDA 11.4, if I use CUDA 10.2, the compute_sm is unrecognizable.

jkjung-avt, 2024-11-19:
Please refer to [TensorRT 7.x.x Release Notes](https://docs.nvidia.com/deeplearning/tensorrt/archives/tensorrt-825/release-notes/tensorrt-7.html):

> The following features are deprecated in TensorRT 7.2.3:
> ......
> We have deprecated the Caffe Parser and UFF Parser in TensorRT 7.0. They will be tested and functional in the next major release of TensorRT 8.0, however, we plan to remove the support in the subsequent major release. Ensure you migrate your workflow to use tf2onnx, keras2onnx or [TensorFlow-TensorRT (TF-TRT)](https://docs.nvidia.com/deeplearning/frameworks/tf-trt-user-guide/index.html) for deployment.

Since the original MTCNN models are in Caffe format (prototxt & caffemodel files), you can no longer use the latest TensorRT library to load those models directly.

One possible workaround is to convert the Caffe model files into ONNX, and then rewrite [create_engines.cpp](https://github.com/jkjung-avt/tensorrt_demos/blob/master/mtcnn/create_engines.cpp) to use ONNX parser instead.

EdithZou, 2024-11-20:
[image]
And the TRT I use is downloaded from the listed packages (TensorRT-7.1.3.4.Ubuntu-18.04.x86_64-gnu.cuda-11.0.cudnn8.0.tar.gz) which supports CUDA11.0.

jkjung-avt, 2024-11-20:
> [TensorRT] WARNING: Half2 support requested on hardware without native FP16 support, performance will be negatively affected.

Are you using a very old GPU which does not support FP16??

Try to disable FP16 (comment out the following line in the code) to see if it resolves the problem.

https://github.com/jkjung-avt/tensorrt_demos/blob/9bc7fd94103976123dcb643bd479e3b6de8b3ecb/yolo/onnx_to_tensorrt.py#L150

EdithZou, 2024-11-20:
> Are you using a very old GPU which does not support FP16??
>
> Try to disable FP16 (comment out the following line in the code) to see if it resolves the problem. [...]

Thx for your reply! I am using Geforce RTX 4060, actually the main problem is:
`[TensorRT] ERROR: ../rtSafe/cuda/caskUtils.cpp (98) - Assertion Error in trtSmToCask: 0 (Unsupported SM.)`
and either by upgrading TRT or upgrading CUDA not work for the problem. I think it is because TensorRT before 8.5.1 don't support compute capability 8.9 (for 4060):
[image]

jkjung-avt, 2024-11-20:
> I think it is because TensorRT before 8.5.1 don't support compute capability 8.9 (for 4060)

I see.