jkjung-avt/tensorrt_demos #120: TypeError: Cannot convert value 0 to a TensorFlow DType.

yqlbu, 2020-05-25:
Hi,

Thank you for contributing such an excellent tutorial. I followed all your step but I encountered the following error when I ran build_engines.sh :

TypeError: Cannot convert value 0 to a TensorFlow DType.

UFF version: 0.6.5
TensorFlow version: 1.14.0

Do you have any ideas what might cause that error?

Thanks,
Kev

[Errors Log]:

Using output node NMS
Converting to UFF graph
Warning: No conversion function registered for layer: NMS_TRT yet.
Converting NMS as custom op: NMS_TRT
WARNING:tensorflow:From /home/kev/archiconda3/envs/trt/lib/python3.6/site-packages/uff/converters/tensorflow/converter.py:179: The name tf.AttrValue is deprecated. Please use tf.compat.v1.AttrValue instead.

Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_conf as custom op: FlattenConcat_TRT
Traceback (most recent call last):
  File "build_engine.py", line 218, in <module>
    main()
  File "build_engine.py", line 201, in main
    debug_mode=False)
  File "/home/kev/archiconda3/envs/trt/lib/python3.6/site-packages/uff/converters/tensorflow/conversion_helpers.py", line 178, in from_tensorflow
    debug_mode=debug_mode)
  File "/home/kev/archiconda3/envs/trt/lib/python3.6/site-packages/uff/converters/tensorflow/converter.py", line 94, in convert_tf2uff_graph
    uff_graph, input_replacements, debug_mode=debug_mode)
  File "/home/kev/archiconda3/envs/trt/lib/python3.6/site-packages/uff/converters/tensorflow/converter.py", line 79, in convert_tf2uff_node
    op, name, tf_node, inputs, uff_graph, tf_nodes=tf_nodes, debug_mode=debug_mode)
  File "/home/kev/archiconda3/envs/trt/lib/python3.6/site-packages/uff/converters/tensorflow/converter.py", line 47, in convert_layer
    return cls.registry_[op](name, tf_node, inputs, uff_graph, **kwargs)
  File "/home/kev/archiconda3/envs/trt/lib/python3.6/site-packages/uff/converters/tensorflow/converter_functions.py", line 21, in convert_placeholder
    dtype = tf2uff.convert_tf2numpy_dtype(tf_node.attr['dtype'].type)
  File "/home/kev/archiconda3/envs/trt/lib/python3.6/site-packages/uff/converters/tensorflow/converter.py", line 103, in convert_tf2numpy_dtype
    return tf.as_dtype(dtype).as_numpy_dtype
  File "/home/kev/archiconda3/envs/trt/lib/python3.6/site-packages/tensorflow_core/python/framework/dtypes.py", line 721, in as_dtype
    (type_value,))
TypeError: Cannot convert value 0 to a TensorFlow DType.

jkjung-avt, 2020-05-25:
Could you try to downgrade your numpy module to see if it resolves the problem?  I'd suggest you to try "numpy==1.16.1".

yqlbu, 2020-05-25:
I downgraded Numpy from 1.18 to 1.16.1, but unfortunately the error still exists.
Have you tried to build the project with UFF v0.6.5? If so, which version of TF should I install?

Thanks

jkjung-avt, 2020-05-25:
tensorflow-1.14.0 should work.

Have you applied [this patch](https://github.com/jkjung-avt/tensorrt_demos/blob/master/ssd/graphsurgeon.patch-4.2.2) to your graphsurgion module?

yqlbu, 2020-05-25:
Yay the problem actually comes from "node_manipulation.py", I followed your suggestion on [issue](https://github.com/jkjung-avt/tensorrt_demos/issues/21), and the issue was resolved. Thanks for your kind help!

I'd like to share my results on AGX Xavier. xD

yqlbu, 2020-05-26:
[image]

I tested the async version script with my AGX Xavier, and it can run @ 70-90 FPS in real-time. However, I wonder is there any way to slow down the processing speed? When the detector ran with a video file input, it seemed like it was actually speeding up the playback.

Thanks,
Kev

jkjung-avt, 2020-05-26:
As far as I know, cv2.VideoCapture.read() does not support rate control on video fie playback.  You probably need to implement rate control (adding proper delays) by yourself.  You could add such code [here](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/camera.py#L217).

closed 2020-05-27 by jkjung-avt