jkjung-avt/tensorrt_demos #122: camera resolution

PatrikPatrik2, 2020-05-27:
I use an on-board camera, and I can change the camera resolution using the -- width 1200 --height 720
It will display a large image on the screen and make object detection, but using 720 and especially 1080 resolution it becomes obvious that the image is not using the defined resolution, instead it seems to upscale an image with low resolution - resulting in a large image with low resolution (large pixels).
Is there some way to correct this?

jkjung-avt, 2020-05-27:
I think this question is not directly related to this repository.  I'd suggest you to post such a question to [NVIDIA Developer Forum](https://forums.developer.nvidia.com/c/agx-autonomous-machines/55).

PatrikPatrik2, 2020-05-27:
OK, thanks

closed 2020-05-27 by PatrikPatrik2