jkjung-avt/tensorrt_demos #124: maximum screen

gpqls, 2020-05-30:
Hey, it's me again.....

I did the whole process perfectly with your code. I carried out #3:SSD.
Is there a way to increase the size of the send screen when you use the webcam to detect in real time?

I wish it was full screen if possible....

jkjung-avt, 2020-05-30:
You could hit 'f' (or 'F') to toggle full-screen when running trt_ssd.py.

The source code is already [there](https://github.com/jkjung-avt/tensorrt_demos/blob/master/trt_ssd.py#L74).

gpqls, 2020-05-31:
Is there a way to change the screen to the full screen without pressing 'f'?

jkjung-avt, 2020-05-31:
https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/display.py#L35

gpqls, 2020-05-31:
I know that part is the part that determines the full screen.  But I don't know how to set the full screen automatically.

jkjung-avt, 2020-05-31:
You just call cvs2.setWindowProperty() as shown in the code.  I expect you to be able to read python source code.

```
    cv2.setWindowProperty(window_name, cv2.WND_PROP_FULLSCREEN,
                          cv2.WINDOW_FULLSCREEN)
```

closed 2020-06-10 by jkjung-avt