jkjung-avt/tensorrt_demos #125: trt-ssd multiple detection boxes

yqlbu, 2020-06-01:
Hi, Mr.Jung

I wanna know are there any ways to solve the multiple detection boxes problem?

I've tried to increase the waitKey value from "1" to "50", and it indeed solved the problem. As a trade-off, I observed that FPS decreased tremendously. Therefore, I wonder is there any other way to achieve the same goal in the meanwhile not decreasing the FPS?

if cv2.waitKey(50) & 0xFF == ord('q'):
            break

Best,
Kevin

jkjung-avt, 2020-06-01:
I guess you are referring to this issue: https://github.com/jkjung-avt/tensorrt_demos/issues/19#issuecomment-597689145.

An alternative is to make copies of the image to avoid inferencing/drawing on the same image over and over again.  More specifically, modify [this line](https://github.com/jkjung-avt/tensorrt_demos/blob/master/trt_ssd.py#L60):

```
        img = cam.read()
```

to

```
        img = cam.read().copy()
```

yqlbu, 2020-06-04:
Thanks for the recommendation, I'd like to try it out :)

closed 2020-06-10 by jkjung-avt