jkjung-avt/tensorrt_demos #127: TypeError: buffer is too small for requested array

Darshcg, 2020-06-03:
While I run: python3 yolov3_to_onnx.py --model yolov3-288
I am getting  error like this:
File "yolov3_to_onnx.py", line 375, in _load_one_param_type buffer=self.weights_file.read(param_size * 4))
TypeError: buffer is too small for requested array

Can you please help to resolve this
Thanks in advance

jkjung-avt, 2020-06-03:
Are you using the code with your own custom trained model?

Darshcg, 2020-06-08:
Yes

Darshcg, 2020-06-08:
And also if I run:  "python3 eval_yolov3.py --model yolov3-288" I am getting TypeError: 'module' object is not callable. Can you please help to resolve the same

jkjung-avt, 2020-06-08:
For your own custom YOLOv3 model, you'll need to modify "category_num" in the code by yourself.  Please refer to https://github.com/jkjung-avt/tensorrt_demos/issues/55 or other similar issues in this repository.

closed 2020-06-10 by jkjung-avt