jkjung-avt/tensorrt_demos #130: ModuleNotFoundError: No module named 'google.protobuf'

yemvibol, 2020-06-07:
I already install protobuf but when I run $ python3 yolov3_to_onnx.py --model yolov3-416 i got the error as below.

Traceback (most recent call last):
  File "yolov3_to_onnx.py", line 59, in <module>
    import onnx
  File "/usr/local/lib/python3.6/dist-packages/onnx/__init__.py", line 8, in <module>
    from onnx.external_data_helper import load_external_data_for_model, write_external_data_tensors
  File "/usr/local/lib/python3.6/dist-packages/onnx/external_data_helper.py", line 10, in <module>
    from .onnx_pb import TensorProto, ModelProto
  File "/usr/local/lib/python3.6/dist-packages/onnx/onnx_pb.py", line 8, in <module>
    from .onnx_pb2 import *  # noqa
  File "/usr/local/lib/python3.6/dist-packages/onnx/onnx_pb2.py", line 7, in <module>
    from google.protobuf.internal import enum_type_wrapper
ModuleNotFoundError: No module named 'google.protobuf'

I tried to set PYTHONPATH=/usr/local/lib/python3.6/dist-packages/protobuf-3.8.0-py3.6-linux-aarch64.egg
but it could not help. 

Please help me if you know the issue.
Thank

jkjung-avt, 2020-06-07:
Try this solution: https://github.com/protocolbuffers/protobuf/issues/6135#issuecomment-546241914

I ever encountered this same problem when I used a virtualenv before.  I fixed it by deleting "__init__.py" in the "google" directory.

yemvibol, 2020-06-09:
Many thanks!! 
Just for the reference, the file is in /usr/local/lib/python3.6/dist-packages/protobuf-3.8.0-py3.6-linux-aarch64.egg/google

closed 2020-06-09 by jkjung-avt