jkjung-avt/tensorrt_demos #131: TypeError: 'module' object is not callable

Darshcg, 2020-06-08:
If I run: "python3 eval_yolov3.py --model yolov3-288" I am getting TypeError: 'module' object is not callable. Can you please help to resolve the same

jkjung-avt, 2020-06-08:
This is too little information.  Please provide the full error log (call trace).

Darshcg, 2020-06-08:
[image]

jkjung-avt, 2020-06-08:
You need to specify "--imgs_dir" and "--annotations" to point to the images and annotations of your own evaluation dataset.  Note the annotations should be in "COCO" format.

https://github.com/jkjung-avt/tensorrt_demos/blob/master/eval_yolov3.py#L31

Darshcg, 2020-06-08:
Actually I am using your Evaluation dataset(https://github.com/jkjung-avt/tensorrt_demos/blob/master/README_eval_ssd.md#preparation). Please help me what to change in line 31.

jkjung-avt, 2020-06-08:
1. Try `sudo pip3 install progressbar2` and run "eval_yolov3.py" again, to see if the error is fixed.

2. Otherwise, you could remove `progressbar()` from the code: https://github.com/jkjung-avt/tensorrt_demos/blob/master/eval_yolov3.py#L53.  That is,

    Replace that line:
    ```
        for jpg in progressbar(jpgs):
    ```
    with:
    ```
        for jpg in jpgs:
    ```

Darshcg, 2020-06-08:
@jkjung-avt, 1st didn't work, and I tried with 2nd, but it is taking so much of time to get the output. Is there any way to speed up(to get the output fast)?

jkjung-avt, 2020-06-08:
The COCO "val2017" contains 5,000 images.  It does take a while to do inference over all these images.  If you don't want to evaluate the model with so many images, you could just delete some jpg files from the "val2017" folder.

Otherwise, if you are evaluating your own custom trained model, you should prepare your own images/annotations instead of using COCO data.

Darshcg, 2020-06-08:
After I reduce the number of images, i get the error as shown below 
[image]

jkjung-avt, 2020-06-08:
I don't have a simple solution.  As stated earlier, you should prepare your own dataset for evaluation.

Darshcg, 2020-06-08:
Thank you very much @jkjung-avt. But I used your model itself.

jkjung-avt, 2020-06-08:
The darknet YOLOv3 COCO model was trained by the original author (not me).  If you're verifying mAP of that model, then it's advised to use all 5,000 images in "val2017".  It might take a few minutes to do inference over all those images.

Darshcg, 2020-06-08:
Even after running Evaluation on 5000 images, then also getting the same error
[image]

jkjung-avt, 2020-06-09:
This is a known issue of pycocotools.  It has been fixed in the latest code.  Check out this discussion for more information: https://github.com/cocodataset/cocoapi/issues/356

Darshcg, 2020-06-09:
How to resolve the issue? I am using NumPy version:1.18.2

jkjung-avt, 2020-06-09:
Install the latest "pycocotools".

Or install from GitHub source:

```
$ sudo pip3 uninstall pycocotools  # remove the old installation
$ cd ${HOME}/src
$ git clone https://github.com/cocodataset/cocoapi.git
$ cd cocoapi/PythonAPI
$ sudo python setup.py build_ext install
```

Darshcg, 2020-06-09:
Thank you very much, Jung. I successfully completed the evaluation.
Btw, what's the different between mAP @ IoU=0.5:0.95 and mAP @ IoU=0.5?

jkjung-avt, 2020-06-09:
https://jkjung-avt.github.io/trt-detection-map/

Darshcg, 2020-06-09:
Thank you very much for all your help and support.

closed 2020-06-10 by jkjung-avt