jkjung-avt/tensorrt_demos #132: Does these support batch inference

silverlining21, 2020-06-11:
https://github.com/jkjung-avt/tensorrt_demos/blob/102193d1f2356949becfb1b450c26e801f10ada0/utils/yolov3.py#L139

f.e.  I have output shape [10, 255, 13, 13], how could it reshape into [13,13,~,85] shape like?

jkjung-avt, 2020-06-12:
I think you could just call _process_yolo_output() on each of the 10 outputs individually.  For example, rewrite [PostprocessYOLO.process()](https://github.com/jkjung-avt/tensorrt_demos/blob/102193d1f2356949becfb1b450c26e801f10ada0/utils/yolov3.py#L106) as something like:

```
    def process(self, outputs, resolution_raw, conf_th):
        ......  # docstring omitted
        batch_size = outputs[0].shape[0]
        for i in range(batch_size):
            outputs_reshaped = list()
            for output in outputs:
                outputs_reshaped.append(self._reshape_output(output[i][None]))

            boxes_xywh, categories, confidences = self._process_yolo_output(
                outputs_reshaped, resolution_raw, conf_th)
            ......  # same code as before
```

Please note that you'd also need to [set max_batch_size to 10](https://github.com/jkjung-avt/tensorrt_demos/blob/102193d1f2356949becfb1b450c26e801f10ada0/yolov3_onnx/onnx_to_tensorrt.py#L71) when building the TensorRT engine.

```
        builder.max_batch_size = 10
```

silverlining21, 2020-06-12:
Got it， thanks a lot.

closed 2020-06-12 by silverlining21