jkjung-avt/tensorrt_demos #133: ValueError: cannot reshape array of size 194688 into shape (1,18,52,52)

bruchpilot, 2020-06-11:
I trained a custom model on windows (one class model, yolov3-416)
All is working fine on this plateform.

After I transfer it on Jetson nano, I got this error when using the model:
ValueError: cannot reshape array of size 194688 into shape (1,18,52,52)

I already changed in utils/yolov3.py the category_num to 1 and the output shapes from [(1, 255 to [(1,18

Where shoud I have a look on to solve my problem?

Thanks

jkjung-avt, 2020-06-12:
Please refer to my new blog post: [TensorRT YOLOv3 For Custom Trained Models](https://jkjung-avt.github.io/trt-yolov3-custom/), as well as the latest code in this repository.

bruchpilot, 2020-06-15:
Hi

Thank you for your post.

I just tested it ou and it is still not working.

Before your update, I did not get error when converting from weights to onyx.

Now, I am getting the following messages :
python3 yolov3_to_onnx.py --model yolov3-416 --category_num 1

Layer of type yolo not supported, skipping ONNX node generation.
Layer of type yolo not supported, skipping ONNX node generation.
Layer of type yolo not supported, skipping ONNX node generation.
Traceback (most recent call last):
  File "yolov3_to_onnx.py", line 826, in <module>
    main()
  File "yolov3_to_onnx.py", line 814, in main
    verbose=True)
  File "yolov3_to_onnx.py", line 439, in build_onnx_graph
    params)
  File "yolov3_to_onnx.py", line 316, in load_conv_weights
    conv_params, 'conv', 'weights')
  File "yolov3_to_onnx.py", line 345, in _create_param_tensors
    conv_params, param_category, suffix)
  File "yolov3_to_onnx.py", line 375, in _load_one_param_type
    buffer=self.weights_file.read(param_size * 4))
TypeError: buffer is too small for requested array

I am becoming totally lost with the transfer of my working model on PC to Jetson.
Do you have an advice for me to get it working.

Thanks a lot for your job.

Christian

jkjung-avt, 2020-06-16:
194688 = 1 * 72 * 52 * 52 = 1 * ( (**19** + 5) * 3 ) * 52 * 52

So I guess the model you've trained is with **19** classes, instead of 1.  Could you double check?  (Try "--category_num 19".)

bruchpilot, 2020-06-16:
I followed this tutorial to build the ONE class yolov3 model https://www.learnopencv.com/training-yolov3-deep-learning-based-custom-object-detector/

When I try it on the PC I used for training, all is working fine.
The trouble comes when I want to use it on the Jetson.
At this moment, I do not know where to look to for problem solving : PC or Jetson … That is the question 😀

jkjung-avt, 2020-06-16:
When you said "all is working find on PC", did you mean that you could optimize your custom trained YOLOv3 model to TensorRT engine and run it on PC?

bruchpilot, 2020-06-16:
I meant that model was working with darknet (Windows 10 with GPU)  and recognized the class I trained it on.

The trouble came on when I tried to transfer the model on jetson and converting it to tensorrt through onnx with your script..

jkjung-avt, 2020-06-16:
Could you provide your model cfg and weights files?  I'll check to see if I could resolve the problem.

bruchpilot, 2020-06-16:
Thank you for your proposal.

I put the files on a private FTP
IP 		88.125.137.95
Port 	16385
User 	IA_public
PWD	Tensor69@

bruchpilot, 2020-06-16:
Sorry, user was wrong.

User -> AI_public

I removed you from the blacklist of the server.

jkjung-avt, 2020-06-16:
I'm still having trouble logging into your FTP server.  Could you use a GoogleDrive or something similar instead?

bruchpilot, 2020-06-16:
This link should do the job

https://fil.email/2rLBdENC?skipreg=true

jkjung-avt, 2020-06-20:
If you compare (diff) your "darknet-yolov3.cfg" against the original "yolov3.cfg", you'll find that your CNN design differs from the original model.  More specifically, you set a different "stride" which would cause the last layer feature map to be 2x2 times bigger than the original design.

```
$ diff $HOME/Downloads/darknet-yolov3.cfg yolov3.cfg
......
716c717
< stride=4
---
> stride=2
......
```

In my opinion, your modified design is not good...

Anyway, if you insist on using this modified design, I think you need to modify the code to accommodate the changed feature map dimension.

Here: https://github.com/jkjung-avt/tensorrt_demos/blob/master/yolov3_onnx/yolov3_to_onnx.py#L804

```
        output_tensor_dims['106_convolutional'] = [c, d //  4, d //  4]
```

and here: https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/yolov3.py#L414

```
                                  (1, filters, h //  4, w //  4)]
```

closed 2020-06-23 by jkjung-avt

bruchpilot, 2020-06-23:
Hello

Your trick worked. I did not thought to look at the cfg file.
I went back to stride=2 and now all is OK.
I am now able to train a model on my PC and transfer it on the Jetson.
Tomorrow will be dedicated to focus on the efficiency of my model on video stream.

You did a very nice job with your work on GitHub.

Thanks a lot

Christian

jkjung-avt, 2020-06-24:
You are welcome.