jkjung-avt/tensorrt_demos #134: yolov3-416.trt no such file or directory

AntoineHoe, 2020-06-13:
Hello,
I met this error while I was testing the dog image and I can't figure it out.
May someone give me an hint ?
Thanks !

[image]

jkjung-avt, 2020-06-14:
Did you follow the steps in [Demo #4: YOLOv3](https://github.com/jkjung-avt/tensorrt_demos#yolov3) and build the TensorRT engine?  After finishing step 3, the file "yolov3_onnx/yolov3-416.trt" should be built.

```
$ python3 onnx_to_tensorrt.py --model yolov3-416
```

AntoineHoe, 2020-06-14:
What an absent-minded I am, I forgot this line.
But when I did, I reach this error and I did see it occur in the other issues :/
[image]

jkjung-avt, 2020-06-14:
Make sure you are using onnx==1.4.1.

```
$ pip3 show onnx
```

AntoineHoe, 2020-06-14:
Yeah, at least I succed this step
[image]

jkjung-avt, 2020-06-15:
https://github.com/onnx/onnx/issues/2534

p.s. I use protobuf-3.8.0.

AntoineHoe, 2020-06-15:
Great, it's working on the dog picture, thanks !

closed 2020-06-15 by AntoineHoe

jkjung-avt, 2022-05-06:
https://github.com/jkjung-avt/tensorrt_demos/issues/539#issuecomment-1119218462