jkjung-avt/tensorrt_demos #135: C++ version of this repository

sleebapaul, 2020-06-14:
What I've done?
-------------------------------

1. I could convert SSD Mobilenet V2 to UFF to `.bin` TensorRT engine using this code base. 
2. I could do the inference and achieved ~20 FPS. 

What is my goal?
------------------------

I would like to rewrite the inference part of this code base to C++.  
That means, I've generated a `.bin` file using this code base, now I want to read the file and do the inference in C++. 

I've managed to compile the C++ project, but on running, I get the following error on de-serializing the `.bin` file. 

```
[06/14/2020-18:04:05] [F] [TRT] Assertion failed: mConcatAxisID == 1 || mConcatAxisID == 2 || mConcatAxisID == 3
flattenConcat.cpp:51
Aborting...
```
 
I would like to know if you can guide me through this issue, and possibly suggest some references to write my C++ version.

closed 2020-06-14 by sleebapaul

jkjung-avt, 2020-06-15:
@sleebapaul  You could resolve the problem by linking your C++ code with "libflattenconcat.so" (the FlattenConcat plugin).

I think you've probably already figured it out since you've closed the issue.

sleebapaul, 2020-06-15:
Linking it with CMake didn't work. Then I used 
` dlopen(<path/to/library>, RTLD_LAZY);`

It worked out. Thanks for the response.

chaitanyarit, 2020-07-01:
@sleebapaul can you please guide me to exact resource which you used to comeup with C++ version. I tried sampleUffSSD.cpp from tensorrt/samples which gave me segfault for ssd_inception_v2 model which I was able to run in python inference.

sleebapaul, 2020-07-01:
I wrote my own version taking this Python API as a reference. You may refer to the following references.

1. https://github.com/eembc/mlmark/
2. https://github.com/dusty-nv/jetson-inference/blob/master/c/tensorNet.cpp
3. https://developer.nvidia.com/blog/speed-up-inference-tensorrt/
4. https://github.com/NVIDIA/TensorRT/blob/master/samples/opensource/sampleUffSSD/sampleUffSSD.cpp