jkjung-avt/tensorrt_demos #136: .bin file in ssd

PythonImageDeveloper, 2020-06-14:
Hi,
**1-** The TensorRT accept three parser : uff / onnx / caffemodel, 
You converted the model in two steps, one is uff and the other is .bin,
Since the TensorRT accept uff parser, we need the uff file, why use then generated .bin file?

**2-** If we use trt library of tensorflow, we can easily pass the frozen model to that api, If we use the last version of that library, Is this easy way to convert the tensort model? If that library be can convert all layers and operation to tensorrt nodes, Is there a difference performance between your way and trt api? 
**3-** Your way is a bit harder, we must to consider plugins and remove nodes and so on, If we use the last version of trt API, that API automatically must to solve this problems, right?

jkjung-avt, 2020-06-15:
1. UFF or ONNX is just an "intermediate" format.  TensorRT can parse those formats but would generate "engines" in its own format.  The "bin" files are "TensorRT engines" serialized into files.

2. TensorRT does not provide a parser to parse tensorflow models directly.  You need to convert your tensorflow model to either UFF or ONNX before optimizing it with TensorRT.

3. Plugins are needed if your model contains layers which are not directly supported by TensorRT.  Using a different API would not help in this case.

4. I don't understand this question.

5. Please read "explicit batch" related parts in TensorRT documentation.

PythonImageDeveloper, 2020-06-15:
1- I want to know, Did you test your ssd codes with batch_size > 1 using TensorRT 5/6? This operation works parallel with this versions?

2- If we have a custom block consists of operations that compatible with uff operations, the only difference is having a custom name for block, Is it easy passed in uff filter, or I need to define a custom plugin for that block?

jkjung-avt, 2020-06-15:
1. For object detection models in our company's products or my own projects, I have only used batch size 1.

2. "Conv2D", "FusedBatchNorm" and "ReLU" are all directly supported by TensorRT.  You could refer to the list [here](https://docs.nvidia.com/deeplearning/tensorrt/support-matrix/index.html#supported-ops).  So you don't need to use plugins for such a custom block.

PythonImageDeveloper, 2020-06-15:
Thanks very much.
Final questions:
1- In your productions in company, you use ssd-mobile-v1-v2 or custom models and then convert to TensorRT?
2- If the model allocated N GB of ram for BS = 1, for BS=2  allocated 2*N GB of ram or smaller?

jkjung-avt, 2020-06-15:
1. We use a custom designed SSD model.  Not ssd_mobilenet_v1 or v2.

2. I don't know.  I don't use batch size 2.

PythonImageDeveloper, 2020-06-15:
Thanks,
Are use tested the ssd_mobile_v3 worked with your codes?

jkjung-avt, 2020-06-16:
I tried to optimize "ssd_mobilenet_v3_coco_(large/small)" to TensorRT through UFF, but did not yet manage to accomplish it.  I put my work in progress in the "ssd_mobilenet_v3" branch: https://github.com/jkjung-avt/tensorrt_demos/tree/ssd_mobilenet_v3

closed 2020-06-23 by jkjung-avt