jkjung-avt/tensorrt_demos #137: Custom model class number problem

AntoineHoe, 2020-06-15:
Dear @jkjung-avt, 

I replace the yolov3-416.weights with my own model (after rename it). Then I tried to build the onnx from it and reach this issue.
[image]
I assume, after some research on a similar issue (#58) that is an class number problem.  I was wondering if the correction was still applying (change the filters lines in the cfg file) If so, every filter that contain 255 or only specific ones ?

thanks for your help !

jkjung-avt, 2020-06-16:
Could you provide your custom yolov3-416.cfg and weights files for checking?  I will take a look.

AntoineHoe, 2020-06-16:
It won't be necessary, I solved it by myself. It was, indeed, a number class pb, I changed the 3 good layers. Anyway,  after that I reached a tensort problem.
[image]
Sounded like the #51 but i solved it by commenting the line `builder.fp16_mode = True`

My current problem is this one : what does mean `pos = np.where(box_class_scores >= conf_th)`
[image]

jkjung-avt, 2020-06-16:
I think this line of code, `pos = np.where(box_class_scores >= conf_th)`, is trying to pick out inference results (corresponding to anchor boxes) with class_scores over the confidence threshold.

According to [this SO post](https://stackoverflow.com/a/37652508/7596504), the warning is likely due to "NaN" (Not a Number, or infinity) outputs by your model.  You could add code to filter those "NaN" values out, or try to train your model better.

By the way, the "weights are out of fp16 range" issue could usually be resolved if you re-train your model with proper "weight decay".

AntoineHoe, 2020-06-16:
Thanks for your answer, yeah the model used on this image was not completely trained. I will try again with a more developped one in few hours.

Concerning the decay, unfortunaly, I do not have the time and experience to solve it by this way. 

Thanks again, you, your repository and your answers

closed 2020-06-16 by AntoineHoe

AntoineHoe, 2020-06-16:
Is there a way to reduce the confidence threshold?