jkjung-avt/tensorrt_demos #140: is there a way to save inference output to a file?

aditjha, 2020-06-19:
i would really like to save the inference output to a specified file, is this possible?

jkjung-avt, 2020-06-20:
Do you mean you'd like to:

* save object detection outputs in COCO (json) or VOC (xml) format?
* or save raw output tensor (numpy array) to a file?
* or else (self-defined format)?

aditjha, 2020-06-25:
I would like to save the resulting input with the object detections, for example if I input a jpeg if its possible to have the model save the jpeg with the bounding boxes to a directory rather than displaying it on the screen. 

But more importantly I also want a way to save/access the coordinates of the bounding boxes on the objects detected. Is there any way that would be possible?

Thank you for your help, it'd really be appreciated :)

jkjung-avt, 2020-06-25:
You could refer to my [eval_ssd.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/eval_ssd.py#L57) as an example of saving detection results as a COCO annotation file.  The saved file is a .json.  You could then use COCO's python API to read the annotations.  Check out [pycocoDemo.ipynb](https://github.com/cocodataset/cocoapi/blob/master/PythonAPI/pycocoDemo.ipynb) as an example.

Otherwise, you could modify the code to save the annotations in Pascal VOC format or another format of your preference.  With Pascal VOC annotation, you would be able to look at it with image labeling tools such as [labelImg](https://github.com/tzutalin/labelImg).

aditjha, 2020-06-25:
okay thank you, i will try to modify the code in trt_yolov3.py to maybe return the 4 bounding box coordinates for each detection on an image and then maybe implement the saving coco results code. 

I was also wondering, if I am mainly interested in the numerical values of the bounding boxes of a detections, is there a simple way to turn off the visualizer (im assuming it would save time as well) or would I just have to comment out the visualization parts of the code.

aditjha, 2020-06-25:
btw thanks for this repo, it seems it will be very useful for running inferences effectively on the robotic cars using an intel camera!

closed 2020-07-02 by jkjung-avt