jkjung-avt/tensorrt_demos #141: yolov3 Segmentation fault in onnx_to_tensor

prathik-naidu, 2020-06-21:
Hi,

I was able to generate a onnx file for my yolov3-spp-1cls.cfg (with custom weights) with some slight modifications to your `yolov3_to_onnx.py` script. Specifically, I modified the `output_tensor_dims` to handle the new category num and also since my input images were of size 256 x 416.

This worked, but I'm getting a segmentation fault when I run `onnx_to_tensorrt.py`. This is what the last few lines of output looks like in verbose mode:

```
[TensorRT] VERBOSE: ModelImporter.cpp:107: Parsing node: 112_convolutional_lrelu [LeakyRelu]
[TensorRT] VERBOSE: ModelImporter.cpp:123: Searching for input: 112_convolutional_bn
[TensorRT] VERBOSE: ModelImporter.cpp:129: 112_convolutional_lrelu [LeakyRelu] inputs: [112_convolutional_bn -> (64, 256, 76, 76)],
[TensorRT] VERBOSE: ImporterContext.hpp:122: Registering layer: 112_convolutional_lrelu for ONNX node: 112_convolutional_lrelu
[TensorRT] VERBOSE: ImporterContext.hpp:97: Registering tensor: 112_convolutional_lrelu for ONNX tensor: 112_convolutional_lrelu
[TensorRT] VERBOSE: ModelImporter.cpp:180: 112_convolutional_lrelu [LeakyRelu] outputs: [112_convolutional_lrelu -> (64, 256, 76, 76)],
[TensorRT] VERBOSE: ModelImporter.cpp:107: Parsing node: 113_convolutional [Conv]
[TensorRT] VERBOSE: ModelImporter.cpp:123: Searching for input: 112_convolutional_lrelu
[TensorRT] VERBOSE: ModelImporter.cpp:123: Searching for input: 113_convolutional_conv_weights
[TensorRT] VERBOSE: ModelImporter.cpp:123: Searching for input: 113_convolutional_conv_bias
[TensorRT] VERBOSE: ModelImporter.cpp:129: 113_convolutional [Conv] inputs: [112_convolutional_lrelu -> (64, 256, 76, 76)], [113_convolutional_conv_weights -> (18, 256, 1, 1)], [113_convolutional_conv_bias -> (18)],
[TensorRT] VERBOSE: builtin_op_importers.cpp:442: Convolution input dimensions: (64, 256, 76, 76)
[TensorRT] VERBOSE: builtin_op_importers.cpp:524: Using kernel: (1, 1), strides: (1, 1), padding: (0, 0), dilations: (1, 1), numOutputs: 18
[TensorRT] VERBOSE: builtin_op_importers.cpp:525: Convolution output dimensions: (64, 18, 76, 76)
[TensorRT] VERBOSE: ImporterContext.hpp:122: Registering layer: 113_convolutional for ONNX node: 113_convolutional
[TensorRT] VERBOSE: ImporterContext.hpp:97: Registering tensor: 113_convolutional for ONNX tensor: 113_convolutional
[TensorRT] VERBOSE: ModelImporter.cpp:180: 113_convolutional [Conv] outputs: [113_convolutional -> (64, 18, 76, 76)],
Segmentation fault (core dumped)
```

Any ideas on why this might be happening or how to go about fixing this issue? Not sure if you tried or were able to get tensorRT working with single class yolov3 cfg.

Thanks so much!

jkjung-avt, 2020-06-21:
This looks like OOM (out of memory).  You could try to use `tegrastats` to monitor system memory usage, and check if that could be the case.

prathik-naidu, 2020-06-21:
I'm actually using an RTX 2080 Ti instead of a Jetson but this should also work – are there any specific modifications that need to be made for this though?

prathik-naidu, 2020-06-22:
Yeah I think something may be wrong with the onnx file. I tried using onnxruntime to run an `ort.InferenceSession` but the error is

```
InvalidArgument: [ONNXRuntimeError] : 2 : INVALID_ARGUMENT : Failed to load model with error: /onnxruntime_src/onnxruntime/core/graph/graph.cc:912 void onnxruntime::Graph::InitializeStateFromModelFileGraphProto() This is an invalid model. Graph output (082_convolutional) does not exist in the graph.
```
Noticed that you define `output_tensor_dims['082_convolutional'] = [c, h // 32, w // 32]` where h and w are my height and width for input images. Any ideas for reasons why the onnx file may not be correct? No error are thrown when running yolo_to_onnx.

jkjung-avt, 2020-06-23:
Could you provide your cfg file?  I can take a look to see if I'd spot something problematic.

prathik-naidu, 2020-06-23:
Sure, below is my cfg in the dropdown (basically identical to the yolov3-1cls cfg from https://github.com/ultralytics/yolov3/blob/master/cfg/yolov3-spp-1cls.cfg)

<details><summary>dropdown with cfg below:</summary>
<p>

```
[net]
# Testing
# batch=1
# subdivisions=1
# Training
batch=64
subdivisions=16
width=608
height=608
channels=3
momentum=0.9
decay=0.0005
angle=0
saturation = 1.5
exposure = 1.5
hue=.1

learning_rate=0.001
burn_in=1000
max_batches = 500200
policy=steps
steps=400000,450000
scales=.1,.1

[convolutional]
batch_normalize=1
filters=32
size=3
stride=1
pad=1
activation=leaky

# Downsample

[convolutional]
batch_normalize=1
filters=64
size=3
stride=2
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=32
size=1
stride=1
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=64
size=3
stride=1
pad=1
activation=leaky

[shortcut]
from=-3
activation=linear

# Downsample

[... 720 lines cut ...]
pad=1
filters=256
activation=leaky

[convolutional]
size=1
stride=1
pad=1
filters=18
activation=linear

[yolo]
mask = 0,1,2
anchors = 10,13,  16,30,  33,23,  30,61,  62,45,  59,119,  116,90,  156,198,  373,326
classes=1
num=9
jitter=.3
ignore_thresh = .7
truth_thresh = 1
random=1

```

</p>
</details>

jkjung-avt, 2020-06-23:
Could you replace these 2 lines in your cfg file:

```
width=608
height=608
```

with this?

```
width=256
height=416
```

Let me know if that fixes the problem.

prathik-naidu, 2020-06-23:
Unfortunately that doesn't fix the problem (still a segmentation fault during parsing):

```
[TensorRT] VERBOSE: ModelImporter.cpp:107: Parsing node: 113_convolutional [Conv]
[TensorRT] VERBOSE: ModelImporter.cpp:123: Searching for input: 112_convolutional_lrelu
[TensorRT] VERBOSE: ModelImporter.cpp:123: Searching for input: 113_convolutional_conv_weights
[TensorRT] VERBOSE: ModelImporter.cpp:123: Searching for input: 113_convolutional_conv_bias
[TensorRT] VERBOSE: ModelImporter.cpp:129: 113_convolutional [Conv] inputs: [112_convolutional_lrelu -> (64, 256, 52, 32)], [113_convolutional_conv_weights -> (18, 256, 1, 1)], [113_convolutional_conv_bias -> (18)],
[TensorRT] VERBOSE: builtin_op_importers.cpp:442: Convolution input dimensions: (64, 256, 52, 32)
[TensorRT] VERBOSE: builtin_op_importers.cpp:524: Using kernel: (1, 1), strides: (1, 1), padding: (0, 0), dilations: (1, 1), numOutputs: 18
[TensorRT] VERBOSE: builtin_op_importers.cpp:525: Convolution output dimensions: (64, 18, 52, 32)
[TensorRT] VERBOSE: ImporterContext.hpp:122: Registering layer: 113_convolutional for ONNX node: 113_convolutional
[TensorRT] VERBOSE: ImporterContext.hpp:97: Registering tensor: 113_convolutional for ONNX tensor: 113_convolutional
[TensorRT] VERBOSE: ModelImporter.cpp:180: 113_convolutional [Conv] outputs: [113_convolutional -> (64, 18, 52, 32)],
Segmentation fault (core dumped)
```

It's parsed most of the model up until here and seems to be segfaulting right after getting the 113_convolutional layer.

prathik-naidu, 2020-06-23:
One issue I noticed is that I was passing in a .pt file instead of .weights in yolo to onnx. I fixed that by converting my .pt file (which I linked in the dropbox above) using: https://github.com/ultralytics/yolov3#darknet-conversion

However, now when I run the yolo to onnx code, I get the error: `TypeError: buffer is too small for requested array`

My cfg (in my previous post) seems to have all the right params and classes, so not sure where there is a mismatch between the .cfg and .weights

jkjung-avt, 2020-06-24:
> I had to make a slight modification to the yolo to onnx script, changing layer_param_block, remainder = remainder.split('\n\n', 1) to: ......  since I was getting this error: ValueError: not enough values to unpack (expected 2, got 1)

This should only happen at the end of the cfg file.  You could get rid of this problem by inserting a blank line at the end of the file.

By the way, this script was originally created by NVIDIA (refer to [this](https://docs.nvidia.com/deeplearning/tensorrt/sample-support-guide/index.html#yolov3_onnx)).  I modified it so that it could run with python3.

jkjung-avt, 2020-06-24:
It's kind of troublesome for me to set up the environment and convert your pt (pytorch weights) file to darknet format.  So I'm not able to run the code with the weights you've provided yet.

But anyway, I think you'll need to modify line #802~804 in yolov3_to_onnx.py.

https://github.com/jkjung-avt/tensorrt_demos/blob/master/yolov3_onnx/yolov3_to_onnx.py#L802

You should set h = 256 and w = 416, and figure out the **correct names of output layers**.  Note the output layers are **not** 082. 094 and 106, since your model contains additional SPP layers.

```
    else:
        output_tensor_dims['???_convolutional'] = [c, h // 32, w // 32]
        output_tensor_dims['???_convolutional'] = [c, h // 16, w // 16]
        output_tensor_dims['???_convolutional'] = [c, h //  8, w //  8]
```

prathik-naidu, 2020-06-24:
Yup, noticed that as well since I'm using yolov3-spp and tried making that change to 088, 100, and 112, however I still get the same error:
```
Traceback (most recent call last):
  File "yolo2onnx.py", line 850, in <module>
    main()
  File "yolo2onnx.py", line 838, in main
    verbose=True)
  File "yolo2onnx.py", line 454, in build_onnx_graph
    params)
  File "yolo2onnx.py", line 322, in load_conv_weights
    conv_params, 'conv', 'weights')
  File "yolo2onnx.py", line 351, in _create_param_tensors
    conv_params, param_category, suffix)
  File "yolo2onnx.py", line 390, in _load_one_param_type
    buffer=buf)
TypeError: buffer is too small for requested array
```

Looking into where it's failing, basically the issue is happening on layer `110_convolutional_conv_weights` where `param_shape = [256, 128, 3, 3]` but the len of the buffer that's being read on`self.weights_file.read(param_size * 4)` is 661500 and not (256 * 128 * 3 * 3 * 4). Not sure why this is happening.

jkjung-avt, 2020-06-24:
The cod in [yolov3_to_onnx.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/yolov3_onnx/yolov3_to_onnx.py#L353) in my repository reads:

```
    def _load_one_param_type(self, conv_params, param_category, suffix):
        """Docstring omitted..."""
        param_name = conv_params.generate_param_name(param_category, suffix)
        channels_out, channels_in, filter_h, filter_w = conv_params.conv_weight_dims
        if param_category == 'bn':
            param_shape = [channels_out]
        elif param_category == 'conv':
            if suffix == 'weights':
                param_shape = [channels_out, channels_in, filter_h, filter_w]
            elif suffix == 'bias':
                param_shape = [channels_out]
        param_size = np.product(np.array(param_shape))
        param_data = np.ndarray(
            shape=param_shape,
            dtype='float32',
            buffer=self.weights_file.read(param_size * 4))
        param_data = param_data.flatten().astype(float)
        return param_name, param_data, param_shape
```

In the code snippet above, "param_size" is directly calculated from "param_shape".  So the read buffer size should match the size of the numpy array.

You seem to be running from another code base (yolo2onnx.py)?

prathik-naidu, 2020-06-24:
I noticed that as well regarding the param_size calculation so not sure why the buffer is much smaller than desired amount being read from the file (param_size * 4). 

I just renamed the file but all the code is the same (made a few modifications to argument handling and model names since I'm working with a custom model/weights, so that shouldn't be a problem. I didn't modify any of the actual conversion code.

If it helps here is a link to the folder with the cfg, weights, and slightly modified yolo2onnx script to replicate this error: https://www.dropbox.com/sh/zh65vm4ci20i8y9/AACG2up4H9iojEdWJHC1fjpaa?dl=0. I'm testing this with the standard yolov3-spp implementation (not my single class version), so I modified the output tensor dims to reflect that with the correct output layers. Ideally, it would be awesome to make the necessary modifications to your code to get it to work with yolov3-spp as well!

jkjung-avt, 2020-06-28:
I think I know the cause of the problem now.  But I might need a few days to implement necessary modifications to the code and to test/verify it.

I think the main issue is bad implementation of [_make_route_node()](https://github.com/jkjung-avt/tensorrt_demos/blob/master/yolov3_onnx/yolov3_to_onnx.py#L660).

jkjung-avt, 2020-06-29:
@prathik-naidu I've put my modified code in the "yolov3-spp" branch.  It could now load the model files you've provided and generate the corresponding TensorRT engine.  Please give it s try:

* Check out my latest code in the yolov3-spp branch: https://github.com/jkjung-avt/tensorrt_demos/tree/yolov3-spp
* Rename your cfg and weights files as: "yolov3-spp-416x256.cfg" and "yolov3-spp-416x256.weights"
* Set "width=416" and "height=256" in "yolov3-spp-416x256.cfg"
* Run the code with "--model yolov3-spp-416x256"

However, the resulting TensorRT engine is not able to detect any object in the test image: dog.jpg.  For that, could you help to verify the weights file?

jkjung-avt, 2020-06-29:
@prathik-naidu Quick update: I use your spp cfg and weights files, but adjust height and width to 416x416.  Then the resulting TensorRT engine seems to detect objects correctly.  So I think my code is behaving correctly.  Please verify it.

prathik-naidu, 2020-06-30:
@jkjung-avt looks good to me! I ended up using torch.onnx.export and needed to make some modifications to my code to help resolve issues but got it working. The changes you made to the branch make sense and looks good to merge in. Thanks for the help!

closed 2020-06-30 by prathik-naidu