jkjung-avt/tensorrt_demos #142: Yolov5 support

rose-jinyang, 2020-06-29:
Hello
How are you?
Thanks for contributing this project.
Could u support to use Yolov5 model(*.pt) in this project?
Thanks

jkjung-avt, 2020-06-29:
Thanks for the suggestion.  I will take a look (but no promise)...

c66tang, 2020-07-11:
> Thanks for the suggestion. I will take a look (but no promise)...

Thanks for sharing your work. I am also looking forward to your yolov5 implementation for a fair comparison.

I tested your "yolov3-416" on TX2 and got ~6fps. I also tried tensorrt_yolov4 from this project (https://github.com/wep21/tensorrt_yolov4_ros). Inference usually took ~330ms for a single image feed from the ros-bridge.

jkjung-avt, 2020-07-18:
I've extended my code to support yolov4 now.  You could refer to the README for inference speed (FPS) numbers I observed on my Jetson Nano DevKit.

I'm not going to work on yolov5 now, since it's based on pytorch (would require too much effort for me).  Let me know if a **darknet implementation of yolov5** is available.  I think I'd be able to tweak the code to support that easily.

closed 2020-07-18 by jkjung-avt

Darshcg, 2021-04-21:
I have the Yolov5 .weights file.
How do I convert to onnx model using your repo?

jkjung-avt, 2021-04-21:
> I have the Yolov5 .weights file.
> How do I convert to onnx model using your repo?

Please provide your cfg file for reference.

Darshcg, 2021-04-21:
Sure @jkjung-avt 
Here it is:
```

# parameters
nc: 80  # number of classes
depth_multiple: 0.33  # model depth multiple
width_multiple: 0.50  # layer channel multiple

# anchors
anchors:
  - [10,13, 16,30, 33,23]  # P3/8
  - [30,61, 62,45, 59,119]  # P4/16
  - [116,90, 156,198, 373,326]  # P5/32

# YOLOv5 backbone
backbone:
  # [from, number, module, args]
  [[-1, 1, Focus, [64, 3]],  # 0-P1/2
   [-1, 1, Conv, [128, 3, 2]],  # 1-P2/4
   [-1, 3, C3, [128]],
   [-1, 1, Conv, [256, 3, 2]],  # 3-P3/8
   [-1, 9, C3, [256]],
   [-1, 1, Conv, [512, 3, 2]],  # 5-P4/16
   [-1, 9, C3, [512]],
   [-1, 1, Conv, [1024, 3, 2]],  # 7-P5/32
   [-1, 1, SPP, [1024, [5, 9, 13]]],
   [-1, 3, C3, [1024, False]],  # 9
  ]

# YOLOv5 head
head:
  [[-1, 1, Conv, [512, 1, 1]],
   [-1, 1, nn.Upsample, [None, 2, 'nearest']],
   [[-1, 6], 1, Concat, [1]],  # cat backbone P4
   [-1, 3, C3, [512, False]],  # 13

   [-1, 1, Conv, [256, 1, 1]],
   [-1, 1, nn.Upsample, [None, 2, 'nearest']],
   [[-1, 4], 1, Concat, [1]],  # cat backbone P3
   [-1, 3, C3, [256, False]],  # 17 (P3/8-small)

   [-1, 1, Conv, [256, 3, 2]],
   [[-1, 14], 1, Concat, [1]],  # cat head P4
   [-1, 3, C3, [512, False]],  # 20 (P4/16-medium)

   [-1, 1, Conv, [512, 3, 2]],
   [[-1, 10], 1, Concat, [1]],  # cat head P5
   [-1, 3, C3, [1024, False]],  # 23 (P5/32-large)

   [[17, 20, 23], 1, Detect, [nc, anchors]],  # Detect(P3, P4, P5)
  ]

```

jkjung-avt, 2021-04-21:
@Darshcg  This is not a DarkNet cfg file.  You'll need to modify the source code by yourself to handle it.