jkjung-avt/tensorrt_demos #144: Output pytorch tensors directly without gpu -> cpu copy

prathik-naidu, 2020-06-30:
Rather than copying the outputs of the tensorRT engine back to cpu (line referenced below), I'd like to output a pytorch cuda tensor that I can directly work with for post processing.

https://github.com/jkjung-avt/tensorrt_demos/blob/dd2471346f5f22894cae925914b769cdc176fbcd/utils/yolov3.py#L383

I referenced code from: https://github.com/NVIDIA-AI-IOT/jetbot/blob/cf3e264ae636d803dcead0549e04b013249f0b30/jetbot/tensorrt_model.py#L102, which binds torch tensors for the input and output and passes those in as bindings to the engine. I tried to replicate this code, by modifying allocate buffers to store a binding to a torch tensor for the output as follows:

```python
# Add host/device mem buffers for input/outputs
if engine.binding_is_input(binding):
    inputs.append(HostDeviceMem(host_mem, device_mem))
    bindings.append(int(device_mem))
else:
    output = torch.empty(size=tuple(shape), dtype=torch.float32, device=torch.device('cuda'))
    outputs.append(output)
    bindings.append(int(output.data_ptr()))
```

However, when I run the code now (which was working fine before), `context.execute_async_v2` throws the error:

`[TensorRT] ERROR: ../rtSafe/cuda/reformat.cu (842) - Cuda Error in NCHWToNCHHW2: 400 (invalid resource handle)`

I have `import pycuda.autoinit` and I did reference #67 and made the following change:

```python
device = cuda.Device(0)
ctx = device.make_context()
self.inputs[0].host_mem = np.ascontiguousarray(imgs)

# run inference function here

ctx.pop()
del ctx
```

However, the same error is still showing up, so I'm wondering if you have any insights on what might be happening? Seems like this is a fairly straightforward change so hopefully you can replicate.

jkjung-avt, 2020-07-01:
Could you try to do the following before calling `context.execute_async_v2()`, as I've done in https://github.com/jkjung-avt/tensorrt_demos/issues/67#issuecomment-604887874?

```
    cuda_ctx.push()  # set this as the active context
```

prathik-naidu, 2020-07-01:
Sorry about that typo in my previous comment but when I add that in, the error is now:

`[TensorRT] ERROR: ../rtSafe/safeContext.cpp (133) - Cudnn Error in configure: 7 (CUDNN_STATUS_MAPPING_ERROR)`

Noticed that my pytorch cuda version `torch.version.cuda` is 10.1 but my tensorRT version is 10.2 so going to fix that and see if it resolves this new issue.

prathik-naidu, 2020-07-01:
Just fixed torch cuda issues so now my torch.version.cuda is 10.2. However, I'm getting the issue from my previous comment `Cudnn Error in configure: 7 (CUDNN_STATUS_MAPPING_ERROR)`. I referenced: https://github.com/NVIDIA/TensorRT/issues/303 and tried moving the engine deserialization and creating the context during the inference code and I'm working with a simple test case:

```python
        cuda.init()
        device = cuda.Device(0)
        ctx = device.make_context()
        ctx.push()

        self.engine = self._load_engine()
        self.context = self.engine.create_execution_context()
        inputs = [torch.ones((1, 3, 256, 416), device="cuda:0")]
        outputs = [torch.zeros((1, 3, 8, 13), device="cuda:0"), torch.zeros((1, 3, 16, 26), device="cuda:0"), 
                    torch.zeros((1, 3, 32, 52), device="cuda:0"), torch.zeros((1, 6552, 6), device="cuda:0")]
        bindings = [_input.data_ptr() for _input in inputs] + [_output.data_ptr() for _output in outputs]
        
        self.context.execute_v2(bindings)

        ctx.pop()
        del ctx
```

Any ideas why this might be happening? I do have a `import pycuda.autoinit` but that doesn't impact anything. One thing to note is that I do have this warning: `[TensorRT] WARNING: TensorRT was linked against cuBLAS 10.2.2 but loaded cuBLAS 10.2.1` but not sure if that's important (I got this before I switched to using pytorch tensors for outputs).

jkjung-avt, 2020-07-02:
I myself have very little experience working with pytorch.  As such, I think I won't be able to offer much help on this issue.

I saw that you've already posted this issue to NVIDIA/TensorRT: https://github.com/NVIDIA/TensorRT/issues/62.  Hopefully you could get the issue resolved from that discussion.

prathik-naidu, 2020-07-02:
Thanks for all the help! If you're curious, posted a solution that ended up working for me (although not entirely sure why) – https://github.com/NVIDIA/TensorRT/issues/62#issuecomment-653085425

closed 2020-07-02 by prathik-naidu

jkjung-avt, 2020-07-02:
> Thanks for all the help! If you're curious, posted a solution that ended up working for me (although not entirely sure why) – NVIDIA/TensorRT#62 (comment)

This is useful information.  Thanks.