jkjung-avt/tensorrt_demos #147: version tag does not match

skywo1f, 2020-07-02:
I am trying to create a trt engine on my pc and then run it on my xavier agx. running “dpkg -l | grep tensorrt” on my pc, I find :
nv-tensorrt-repo-ubuntu1804-cuda10.0-trt6.0.1.5-ga-20190913
on my xavier agx, I find:
6.0.1.10-1+cuda10.0

but when I try to run my trt engine on the xavier, it tells me serialization error in verify header: 0 (version tag does not match).
what am I doing wrong?

jkjung-avt, 2020-07-03:
You cannot create a TensorRT engine on an x86_64 PC and try to do inference with the engine on the Jetson AGX Xavier.  In other words, the TensorRT engine should be built and serialized on the exact same target platform.

skywo1f, 2020-07-06:
1. This is what I used to do. But then nvidia updates happened and now it doesnt work. 
2. The xavier agx doesnt have the tensor rt capabilities to make a trt engine
3. even yolov3_to_onnx.py doesnt work : buffer is too small for requested array, 1.0 has type numpy.float32, but expected one of int,long, float

skywo1f, 2020-07-06:
the full error is:
~/tensorrt_demos$ python3 trt_yolov3.py --model yolov3-608 --vid 0 --usb
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
[TensorRT] ERROR: INVALID_CONFIG: The engine plan file is generated on an incompatible device, expecting compute 7.2 got compute 7.5, please rebuild.
[TensorRT] ERROR: engine.cpp (1324) - Serialization Error in deserialize: 0 (Core engine deserialization failure)
[TensorRT] ERROR: INVALID_STATE: std::exception
[TensorRT] ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.
Traceback (most recent call last):
File “trt_yolov3.py”, line 112, in
main()
File “trt_yolov3.py”, line 98, in main
trt_yolov3 = TrtYOLOv3(args.model, (h, w), args.category_num)
File “/home/xavier/tensorrt_demos/utils/yolov3.py”, line 447, in init
self.context = self._create_context()
File “/home/xavier/tensorrt_demos/utils/yolov3.py”, line 399, in _create_context
return self.engine.create_execution_context()
AttributeError: ‘NoneType’ object has no attribute ‘create_execution_context’
Exception ignored in: <bound method TrtYOLOv3.del of <utils.yolov3.TrtYOLOv3 object at 0x7f7f840fd0>>
Traceback (most recent call last):
File “/home/xavier/tensorrt_demos/utils/yolov3.py”, line 455, in del
del self.stream
AttributeError: stream

skywo1f, 2020-07-06:
ok, I solved the first issue ("1.0 has type numpy.float32, but expected one of int,long, float"). Turns out python3 was using the wrong old version of protobuf, but when I tried to update it, it would see the python2.7 version and think it was up to date. Deleting the python3 protobuf and reinstalling updated it.

But I am still having onnx problems. 1.7 does not work, neither does 1.2.2. Any ideas which version to try?

skywo1f, 2020-07-06:
my second issue:"buffer is too small for requested array " was caused by a weights/cfg files missmatch from the automatic downloading script. Getting them from the yolo website fixed this problem

closed 2020-07-10 by jkjung-avt