jkjung-avt/tensorrt_demos #148: About context.pop()

1003581, 2020-07-03:
Hello,I create gpu context as follow,

Idevice = cuda.Device(0)
context = device.make_context()

//trt inferencing

context.pop()
del context

but " context.pop() " dose not work，return “”PyCUDA ERROR: The context stack was not empty upon module cleanup. “.  
If I ues  context.detach()，it work. I have not idea about this situation. Do you konw why?

jkjung-avt, 2020-07-03:
Maybe you could read [PyCUDA documentation](https://documen.tician.de/pycuda/driver.html#devices-and-contexts).

jkjung-avt, 2020-07-23:
> but " context.pop() " does not work，return “PyCUDA ERROR: The context stack was not empty upon module cleanup. “.

Please check whether you have done additional context.push()'s in the code.  I guess you need to do 1 pop() for each push() you've called.

closed 2020-07-23 by jkjung-avt