jkjung-avt/tensorrt_demos #149: [Demo3 SSD] nmsPlugin.cpp assertion error

Ragavendrams, 2020-07-06:
Hi, 
I am using the demo 3 to convert my SSD model from Tensorflow to TensorRT. I have changed the number of classes in the model from 91 to 2 (changed also the build_engine file). However, when I run buildengine.py, I get an assertion error in the nmsplugin. I guess this is because of the input order. However even after checking all 6 combinations, the error persists. Can you let me know how to resolve this?

**Config**:
Tensorflow 1.15.2
Jetpack 4.4 DP
Jetson TX2

**File:**

[ssd_mobilenet_v2_coco.zip](https://github.com/jkjung-avt/tensorrt_demos/files/4878350/ssd_mobilenet_v2_coco.zip)

Thanks in advance!
- Ragavendra

joelosw, 2020-07-06:
Can you specify which assertion error you get? 
Line 246 or 249?

The problem might be, that your models feature map shapes do not match the specified feature map shape in the plugin, which results in a mismatch of generated anchor boxes and predictions. (At least I had that problem once....)

Ragavendrams, 2020-07-06:
hi,
Thanks for the quick reply.
 it is in line 246. how do i check the feature map shapes?

joelosw, 2020-07-06:
Ahh, thanks for adding the output!
What is the image size that you are using in your code and for the model?

Ragavendrams, 2020-07-06:
The original model input size is 300x300 and I have not changed that anywhere.

joelosw, 2020-07-06:
Mhm, okay. Could you maybe provide the config file, that you used to train your model (assuming that you used the tf Object Detection API)

Ragavendrams, 2020-07-06:
Sure. Here it is.

[pipeline.zip](https://github.com/jkjung-avt/tensorrt_demos/files/4878467/pipeline.zip)

joelosw, 2020-07-06:
Okay, another guess: 
You specified two classes in your pipeline. So I guess you have two classes + background. 
Did you set num_classes in the build_engine.py to 3? At this place you count the background class in.

Ragavendrams, 2020-07-06:
Perfect! That did the trick. Thanks a lot.
However, if i were to change the input size, say to 250x250, what else should i change?

joelosw, 2020-07-06:
Great!
So if you have a model trained on a different input size, you need to change the featureMapShapes in the GridAnchorGenerator Plugin. 

I personally use this function to calculate the feature_map_shapes: 
`def get_feature_map_shape(img_width):
    fms = []
    curr = int(np.ceil(img_width / 16.0))
    for i in range(6):
        fms.append(curr)
        curr = int(np.ceil(curr / 2.0))
    return fms`

But I cannot guarantee that it works for every combination. You could also manually look the shapes up by exploring the graph with tensorboard. 

And of course you would have to change the INPUT_DIMS variable in the file.

Ragavendrams, 2020-07-14:
Thanks a lot for your help.  
I am able to train the models with different input sizes now.
Though I was not able to view the feature map shapes using Tensorboard. Only the channel information is available. 
It is of form ?x?x?xC.  I guess this is because in tensorboard the input shape is not known. Is there a way to specify that?

joelosw, 2020-07-14:
It depends on how you visualize your graph in tensorboard. 
You will not be able to see the sizes from the frozen_inference_graph. 
But in my case, when run tensorboard on the evaluation_dir or train_dir where I trained/evaluated the model, it shows me the dimensions.

Ragavendrams, 2020-07-14:
Yes I can see that too now.  
Thanks again for all your help. : )
I will close this issue now.

closed 2020-07-14 by Ragavendrams

wadhwasahil, 2022-05-17:
I am also having similar issue. #assertion/opt/tensorrt/TensorRT/plugin/nmsPlugin/nmsPlugin.cpp,246

My image size is 300 * 300. I also tried various combinations of **'input_order**' aka l`oc_data, conf_data, priorbox_data` but to no avail

jkjung-avt, 2022-05-17:
NVIDIA has open-sourced the plugin: https://github.com/NVIDIA/TensorRT/blob/main/plugin/nmsPlugin/nmsPlugin.cpp

You might need to refer to a older version of this source file, depending on which version of TensorRT you are using.

wadhwasahil, 2022-05-17:
@jkjung-avt I have opened a similar issue here https://github.com/jkjung-avt/tensorrt_demos/issues/553

Please let me know what I am doing wrong