jkjung-avt/tensorrt_demos #151: YOLOV3，value Error

Water2style, 2020-07-13:
Hi jkjung:

Note: I used the yolov3.weights and .cfg from darknet site.  i didn't use your download_yolov3.sh.  Is this the problem? 

Traceback (most recent call last):
  File "trt_yolov3.py", line 112, in <module>
    main()
  File "trt_yolov3.py", line 104, in main
    loop_and_detect(cam, trt_yolov3, conf_th=0.3, vis=vis)
  File "trt_yolov3.py", line 57, in loop_and_detect
    boxes, confs, clss = trt_yolov3.detect(img, conf_th)
  File "/home/nvidia/Desktop/TensorRT/tensorrt_demos/utils/yolov3.py", line 478, in detect
    in zip(trt_outputs, self.output_shapes)]
  File "/home/nvidia/Desktop/TensorRT/tensorrt_demos/utils/yolov3.py", line 477, in <listcomp>
    trt_outputs = [output.reshape(shape) for output, shape
ValueError: cannot reshape array of size 92055 into shape (1,255,13,13)

jkjung-avt, 2020-07-13:
My code relies on filename to determine dimensions of the input image.  So please rename ["yolov3.cfg", "yolov3.weigths" and "yolov3.trt"] to ["yolov3-608.cfg", "yolov3-608.weights" and "yolov3-608.trt"] (or "yolov3-608x608.*").

Also, please make sure the "width=608" and "height=608" inside the cfg file match the filename.

Water2style, 2020-07-13:
> My code relies on filename to determine dimensions of the input image. So please rename ["yolov3.cfg", "yolov3.weigths" and "yolov3.trt"] to ["yolov3-608.cfg", "yolov3-608.weights" and [...]

Thank you ,608 works well~   but when i Change the 608 to 416...it doesn't work

jkjung-avt, 2020-07-13:
For that, you need to rename the cfg and weights to 416, re-run yolov3_to_onnx and onnx_to_tensorrt scripts.

closed 2020-07-18 by jkjung-avt