jkjung-avt/tensorrt_demos #152: Does this inference support yolov4 .engine file?

saikrishnadas, 2020-07-15:
Hey thanks for your contribution,
I would like to know if this inference or test code works with my yolov4 model?
I have a custom model, that was trained and converted to .engine file converted on jetson xavier nx. 
Does your test code or inference code work with that .engine file?

Hoping to hear from you soon. @jkjung-avt jk

jkjung-avt, 2020-07-15:
I'm currently developing/modifying the code to support YOLOv4.  It is a work in progress and is currently on the "yolov4" branch: https://github.com/jkjung-avt/tensorrt_demos/tree/yolov4.

As of now, the modified code is able to parse YOLOv4 cfg/weights and generate a TensorRT engine.  But since TensorRT does not natively support "Mish" activation, I replace "Mish" with "Leaky ReLu" for now.  And the inference result is a little bit off.

You could check out the current code if you'd like.  Otherwise, I plan to write a blog post about TensorRT YOLOv4 when I finish implementing it.

saikrishnadas, 2020-07-15:
Yeah cool, 
But I managed to convert the model into .engine, I would like to know if your code for testing(not conversion ) will work on this?
```

python3 trt_yolov3.py --model yolov3-416
                        --image --filename ${HOME}/Pictures/dog.jpg
```
Thanks :)

jkjung-avt, 2020-07-15:
If your TensorRT engine exactly matches YOLOv4 architecture and does not require any plugin, then you could use trt_yolo.py for inference.  (Note that I have renamed trt_yolov3.py in the new branch.)

But as explained earlier, YOLOv4 was trained with "Mish" activations.  Inference results are slightly different when I replaced all "Mish" with "Leaky ReLu".

saikrishnadas, 2020-07-16:
Cool,
It's best for me to wait until your repo is done :)

Thanks

marvision-ai, 2020-07-16:
@jkjung-avt If you are using LeakyRelu then I assume your code should work with Yolov4-tiny. Correct? That model is not trained with Mish.

jkjung-avt, 2020-07-16:
I have solved the problem of Mish activation today: [Implement 'Mish' activation with 'Softplus' + 'Tanh' + 'Mul'](https://github.com/jkjung-avt/tensorrt_demos/commit/7cbb8b5d239e6091c8b3dcc9602456f438069381).  So the current code in my "yolov4" branch could build TensorRT engines for yolov4-288/yolov4-416/yolov4-608 models and seems to produce correct inference results with the engines.

My code is still having problems with yolov4-tiny models.  I'll look into that tomorrow.

marvision-ai, 2020-07-16:
@jkjung-avt  Oh great! Good work.

I hope you can solve the tiny-models (including 3 layer tiny models) because they are providing amazing accuracy at amazing speeds.

jkjung-avt, 2020-07-18:
I have merged my "yolov4" branch into master.  The current code has been verified to work well for yolov4-288, yolov4-416, and yolov4-608 models.

I will close this issue for now.  And I shall continue working on yolov4-tiny models.

closed 2020-07-18 by jkjung-avt