jkjung-avt/tensorrt_demos #156: Problem with SSP custom conversion model

adrianosantospb, 2020-07-20:
Hello, 

Tks for share your work to us. I'm trying to convert a YOLOV3 SSP model, but I've got this error:

` onnx.checker.check_model(yolo_model_def)
  File "/home/workstation/anaconda3/envs/tensorrtdemo/lib/python3.7/site-packages/onnx/checker.py", line 93, in check_model
    C.check_model(model.SerializeToString())
onnx.onnx_cpp2py_export.checker.ValidationError: Op registered for Upsample is deprecated in domain_version of 12

==> Context: Bad node spec: input: "092_convolutional_lrelu" input: "093_upsample_scale" output: "093_upsample" name: "093_upsample" op_type: "Upsample" attribute { name: "mode" s: "nearest" type: STRING }`

Do you have some ideia?

wjshamblin, 2020-07-20:
Make sure that you have onnx 1.4.1 installed. That will probably fix your issue.

jkjung-avt, 2020-07-21:
@adrianosantospb  I concur with wjshamblin.  The code in this repository is compatible with "onnx==1.4.1" (opset 10).  Your error message shows that you are using opset 12.

So please downgrade your onnx module.

```
$ sudo pip3 install onnx==1.4.1
```

adrianosantospb, 2020-07-22:
@jkjung-avt and @wjshamblin , it works. Tks. But now, when I'm trying to convert to tensorrt I've got the **segment fault core dumped** error.

jkjung-avt, 2020-07-23:
Try running "onnx_to_tensorrt.py" with "--verbose" to see if you could get some useful error logs.

Otherwise, could you provide the full output of "yolo_to_onnx.py"?

LorenzoMonti, 2020-07-23:
@jkjung-avt i have the same error of @adrianosantospb and attached you can find my log. Any suggestions?

Thank you in advance
[onnx_to_tensorrt_log.log](https://github.com/jkjung-avt/tensorrt_demos/files/4965281/onnx_to_tensorrt_log.log)

jkjung-avt, 2020-07-23:
@LorenzoMonti  I cannot spot anything in the onnx_to_tensorrt log.  Could you also provide your cfg file and "yolo_to_onnx" log?

adrianosantospb, 2020-07-23:
[net]
# Testing
# batch=1
# subdivisions=1
# Training
batch=16
subdivisions=1
width=416
height=416
channels=3
momentum=0.9
decay=0.0005
angle=0
saturation = 1.5
exposure = 1.5
hue=.1

learning_rate=0.001
burn_in=1000
max_batches = 120200
policy=steps
steps=70000,100000
scales=.1,.1

[convolutional]
batch_normalize=1
filters=32
size=3
stride=1
pad=1
activation=leaky

# Downsample

[convolutional]
batch_normalize=1
filters=64
size=3
stride=2
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=32
size=1
stride=1
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=64
size=3
stride=1
pad=1
activation=leaky

[shortcut]
from=-3
activation=linear

# Downsample

[convolutional]
batch_normalize=1
filters=128
size=3
stride=2
pad=1
[... 757 lines cut ...]

[convolutional]
batch_normalize=1
size=3
stride=1
pad=1
filters=256
activation=leaky

[convolutional]
size=1
stride=1
pad=1
filters=18
activation=linear

[yolo]
mask = 0,1,2
anchors = 10,13,  16,30,  33,23,  30,61,  62,45,  59,119,  116,90,  156,198,  373,326
classes=1
num=9
jitter=.3
ignore_thresh = .7
truth_thresh = 1
random=1

adrianosantospb, 2020-07-23:
Yes. https://notepad.pw/go152u3a

adrianosantospb, 2020-07-23:
In my case, I didn't have problem to convert Yolo to Onnx. No logs, no errors, no messages. 
My cfg file is: https://notepad.pw/go152u3a

LorenzoMonti, 2020-07-23:
@jkjung-avt you're right, me neither spot anything. Attached the cfg file.

LorenzoMonti, 2020-07-23:
I switch the training batch and subdivision to the test one and it works for me!

jkjung-avt, 2020-07-23:
@LorenzoMonti @adrianosantospb  I need to see the "yolo_to_onnx" log, in order to verify whether number of output channels of each layer has been calculated correctly.  Otherwise, I'll need to ask for your trained weights to debug the problem...

adrianosantospb, 2020-07-23:
YOLO_TO_ONNX log

416
Parsing DarkNet cfg file...
Building ONNX graph...
graph yolov3-ssp-416 (
  %000_net[FLOAT, 1x3x416x416]
) initializers (
  %001_convolutional_bn_scale[FLOAT, 32]
  %001_convolutional_bn_bias[FLOAT, 32]
  %001_convolutional_bn_mean[FLOAT, 32]
  %001_convolutional_bn_var[FLOAT, 32]
  %001_convolutional_conv_weights[FLOAT, 32x3x3x3]
  %002_convolutional_bn_scale[FLOAT, 64]
  %002_convolutional_bn_bias[FLOAT, 64]
  %002_convolutional_bn_mean[FLOAT, 64]
  %002_convolutional_bn_var[FLOAT, 64]
  %002_convolutional_conv_weights[FLOAT, 64x32x3x3]
  %003_convolutional_bn_scale[FLOAT, 32]
  %003_convolutional_bn_bias[FLOAT, 32]
  %003_convolutional_bn_mean[FLOAT, 32]
  %003_convolutional_bn_var[FLOAT, 32]
  %003_convolutional_conv_weights[FLOAT, 32x64x1x1]
  %004_convolutional_bn_scale[FLOAT, 64]
  %004_convolutional_bn_bias[FLOAT, 64]
  %004_convolutional_bn_mean[FLOAT, 64]
  %004_convolutional_bn_var[FLOAT, 64]
  %004_convolutional_conv_weights[FLOAT, 64x32x3x3]
  %006_convolutional_bn_scale[FLOAT, 128]
  %006_convolutional_bn_bias[FLOAT, 128]
  %006_convolutional_bn_mean[FLOAT, 128]
  %006_convolutional_bn_var[FLOAT, 128]
  %006_convolutional_conv_weights[FLOAT, 128x64x3x3]
  %007_convolutional_bn_scale[FLOAT, 64]
  %007_convolutional_bn_bias[FLOAT, 64]
  %007_convolutional_bn_mean[FLOAT, 64]
  %007_convolutional_bn_var[FLOAT, 64]
  %007_convolutional_conv_weights[FLOAT, 64x128x1x1]
  %008_convolutional_bn_scale[FLOAT, 128]
  %008_convolutional_bn_bias[FLOAT, 128]
  %008_convolutional_bn_mean[FLOAT, 128]
  %008_convolutional_bn_var[FLOAT, 128]
  %008_convolutional_conv_weights[FLOAT, 128x64x3x3]
  %010_convolutional_bn_scale[FLOAT, 64]
  %010_convolutional_bn_bias[FLOAT, 64]
  %010_convolutional_bn_mean[FLOAT, 64]
  %010_convolutional_bn_var[FLOAT, 64]
  %010_convolutional_conv_weights[FLOAT, 64x128x1x1]
  %011_convolutional_bn_scale[FLOAT, 128]
  %011_convolutional_bn_bias[FLOAT, 128]
  %011_convolutional_bn_mean[FLOAT, 128]
  %011_convolutional_bn_var[FLOAT, 128]
  %011_convolutional_conv_weights[FLOAT, 128x64x3x3]
  %013_convolutional_bn_scale[FLOAT, 256]
  %013_convolutional_bn_bias[FLOAT, 256]
  %013_convolutional_bn_mean[FLOAT, 256]
  %013_convolutional_bn_var[FLOAT, 256]
  %013_convolutional_conv_weights[FLOAT, 256x128x3x3]
  %014_convolutional_bn_scale[FLOAT, 128]
  %014_convolutional_bn_bias[FLOAT, 128]
  %014_convolutional_bn_mean[FLOAT, 128]
  %014_convolutional_bn_var[FLOAT, 128]
  %014_convolutional_conv_weights[FLOAT, 128x256x1x1]
  %015_convolutional_bn_scale[FLOAT, 256]
  %015_convolutional_bn_bias[FLOAT, 256]
  %015_convolutional_bn_mean[FLOAT, 256]
  %015_convolutional_bn_var[FLOAT, 256]
  %015_convolutional_conv_weights[FLOAT, 256x128x3x3]
  %017_convolutional_bn_scale[FLOAT, 128]
  %017_convolutional_bn_bias[FLOAT, 128]
[... 553 lines cut ...]
  %114_convolutional = Conv[auto_pad = 'SAME_LOWER', dilations = [1, 1], kernel_shape = [3, 3], strides = [1, 1]](%113_convolutional_lrelu, %114_convolutional_conv_weights)
  %114_convolutional_bn = BatchNormalization[epsilon = 9.99999974737875e-06, momentum = 0.990000009536743](%114_convolutional, %114_convolutional_bn_scale, %114_convolutional_bn_bias, %114_convolutional_bn_mean, %114_convolutional_bn_var)
  %114_convolutional_lrelu = LeakyRelu[alpha = 0.100000001490116](%114_convolutional_bn)
  %115_convolutional = Conv[auto_pad = 'SAME_LOWER', dilations = [1, 1], kernel_shape = [1, 1], strides = [1, 1]](%114_convolutional_lrelu, %115_convolutional_conv_weights)
  %115_convolutional_bn = BatchNormalization[epsilon = 9.99999974737875e-06, momentum = 0.990000009536743](%115_convolutional, %115_convolutional_bn_scale, %115_convolutional_bn_bias, %115_convolutional_bn_mean, %115_convolutional_bn_var)
  %115_convolutional_lrelu = LeakyRelu[alpha = 0.100000001490116](%115_convolutional_bn)
  %116_maxpool = MaxPool[auto_pad = 'SAME_UPPER', kernel_shape = [5, 5], strides = [1, 1]](%115_convolutional_lrelu)
  %118_maxpool = MaxPool[auto_pad = 'SAME_UPPER', kernel_shape = [9, 9], strides = [1, 1]](%115_convolutional_lrelu)
  %120_maxpool = MaxPool[auto_pad = 'SAME_UPPER', kernel_shape = [13, 13], strides = [1, 1]](%115_convolutional_lrelu)
  %121_route = Concat[axis = 1](%120_maxpool, %118_maxpool, %116_maxpool, %115_convolutional_lrelu)
  %122_convolutional = Conv[auto_pad = 'SAME_LOWER', dilations = [1, 1], kernel_shape = [3, 3], strides = [1, 1]](%121_route, %122_convolutional_conv_weights)
  %122_convolutional_bn = BatchNormalization[epsilon = 9.99999974737875e-06, momentum = 0.990000009536743](%122_convolutional, %122_convolutional_bn_scale, %122_convolutional_bn_bias, %122_convolutional_bn_mean, %122_convolutional_bn_var)
  %122_convolutional_lrelu = LeakyRelu[alpha = 0.100000001490116](%122_convolutional_bn)
  %123_convolutional = Conv[auto_pad = 'SAME_LOWER', dilations = [1, 1], kernel_shape = [1, 1], strides = [1, 1]](%122_convolutional_lrelu, %123_convolutional_conv_weights)
  %123_convolutional_bn = BatchNormalization[epsilon = 9.99999974737875e-06, momentum = 0.990000009536743](%123_convolutional, %123_convolutional_bn_scale, %123_convolutional_bn_bias, %123_convolutional_bn_mean, %123_convolutional_bn_var)
  %123_convolutional_lrelu = LeakyRelu[alpha = 0.100000001490116](%123_convolutional_bn)
  %124_convolutional = Conv[auto_pad = 'SAME_LOWER', dilations = [1, 1], kernel_shape = [3, 3], strides = [1, 1]](%123_convolutional_lrelu, %124_convolutional_conv_weights)
[... cut ...] scale, %124_convolutional_bn_bias, %124_convolutional_bn_mean, %124_convolutional_bn_var)
  %124_convolutional_lrelu = LeakyRelu[alpha = 0.100000001490116](%124_convolutional_bn)
  %125_convolutional = Conv[auto_pad = 'SAME_LOWER', dilations = [1, 1], kernel_shape = [1, 1], strides = [1, 1]](%124_convolutional_lrelu, %125_convolutional_conv_weights, %125_convolutional_conv_bias)
  return %082_convolutional, %094_convolutional, %106_convolutional
}
Checking ONNX model...
Saving ONNX file...
Done.

jkjung-avt, 2020-07-24:
@adrianosantospb  Thanks.  By looking at your "yolo_to_onnx" log, you could easily see the "output" layers (convolution layers with 18 channels output) are 89, 107, and 125.

So you need to modify the code [here](https://github.com/jkjung-avt/tensorrt_demos/blob/031bc4e4da2b0a7f85dd221802fcd2b595f269cc/yolo/yolo_to_onnx.py#L911) and convert onnx/tensorrt again.

```
        elif 'spp' in args.model:
            output_tensor_dims['089_convolutional'] = [c, h // 32, w // 32]
            output_tensor_dims['107_convolutional'] = [c, h // 16, w // 16]
            output_tensor_dims['125_convolutional'] = [c, h //  8, w //  8]
```

adrianosantospb, 2020-07-24:
Hello, @jkjung-avt . Tks for your support. So, when I define the category_num, isn't it automatic?  -- c = (args.category_num + 5) * 3
In my case as you know, I just have 1 category and I did the calculation to filters=[5 + n] * 3.

adrianosantospb, 2020-07-24:
I ran the script using the default weight and cfg file from the download_yolov3.sh file. I did the convertion to onnx and trt without problems. But I did a test using this weight here yolov3-ssp.weights (https://drive.google.com/drive/folders/1LezFG5g3BCW6iYaV89B2i64cqEUZD7e0) and https://github.com/ultralytics/yolov3/blob/master/cfg/yolov3-spp.cfg. I only added the image size (416) and I got the Segmentation fault (core dumped) message problem. It seems the script has some problem to convert SPP architecture.

jkjung-avt, 2020-07-24:
> So, when I define the category_num, isn't it automatic? -- c = (args.category_num + 5) * 3
In my case as you know, I just have 1 category and I did the calculation to filters=[5 + n] * 3.

Yes.  The calculation is already implemented in the code: [here](https://github.com/jkjung-avt/tensorrt_demos/blob/50895c73f7362dba0750fa38dc89e637b658992f/yolo/yolo_to_onnx.py#L906) and [here](https://github.com/jkjung-avt/tensorrt_demos/blob/50895c73f7362dba0750fa38dc89e637b658992f/utils/yolo.py#L412).

I previously tested the code with the [yolov3-spp weights provided by prathik-naidu](https://github.com/jkjung-avt/tensorrt_demos/issues/141#issuecomment-648917818).  I thought it worked OK.

I will download your cfg/weights and check it tomorrow.

adrianosantospb, 2020-07-24:
@jkjung-avt , tks a lot. I will on tomorrow as well. If you think useful, I could share the weights that I'm using (only one class) and the cfg file, and I could help you, if you need.

adrianosantospb, 2020-07-24:
So... I did some tests. I've used an old model that I've trained with the classical YOLOV3 arch. This model also was trained to detect only one class. I've got to convert it to ONNX and to TRT.

adrianosantospb, 2020-07-24:
Using the trt file, I've tried to test using this script:

`import cv2
import pycuda.autoinit  # This is needed for initializing CUDA driver

from yolo_classes import get_cls_dict
from yolo import TrtYOLO

conf_th = 0.3 
img = "./image_example.jpg"

img = cv2.imread(img)

(h, w) = img.shape[:2]

trt_yolo = TrtYOLO("yolov3-416", (h, w), 1)

boxes, confs, clss = trt_yolo.detect(img, conf_th)

for box in boxes:
    print(box)`

I've got the follow error:

pycuda._driver.LogicError: cuMemcpyHtoDAsync failed: invalid argument

jkjung-avt, 2020-07-25:
You just need to replace this:

```
trt_yolo = TrtYOLO("yolov3-416", (h, w), 1)
```

with:

```
trt_yolo = TrtYOLO("yolov3-416", (416, 416), 1)
```

adrianosantospb, 2020-07-25:
Tks, @jkjung-avt . This image is 416x416 (or it should be... ) Well, no error message, but I didn't get detection... It sounds strange to me. This model isn't the best I have, but it is enough accurate. Do you think that some problems happened during the conversion?

jkjung-avt, 2020-07-25:
Try to lower "conf_th" (try 0.1 or even lower) to see if you get any detection at all.

adrianosantospb, 2020-07-25:
Yes. I've changed the conf_th to 0.01 and I've seen the detection. "Good news", dont you? Despite the low conf_th, I didn't see so many problems for this model. I look forward to the SSP working too.

jkjung-avt, 2020-07-26:
@adrianosantospb  I cannot reproduce the problem with the default yolov3-spp cfg/weights you've provided.  Here's what I tested.

1. Download "yolov3-spp.cfg" from https://github.com/ultralytics/yolov3/blob/master/cfg/yolov3-spp.cfg
2. Download "yolov3-spp.weights" from https://drive.google.com/drive/folders/1LezFG5g3BCW6iYaV89B2i64cqEUZD7e0
3. Set `width=416` and `height=416` and rename "yolov3-spp.cfg" to "yolov3-spp-416.cfg"
4. Rename "yolov3-spp.weights" to "yolov3-spp-416.weights"
5. Run "yolo_to_onnx.py" and "onnx_to_tensorrt.py" to convert the model to TensorRT engine
6. Test the TensorRT engine with "trt_yolo.py"

I don't see any problem (no core dump) when running "onnx_to_tensorrt.py".  And the inference results seem correct.

Could you double check?

adrianosantospb, 2020-07-27:
Him @jkjung-avt . Tks for your reply. 
Well, I did the test here and I've got the same problem "Segmentation fault (core dumped)". 
The steps 1, 2, 3, 4 and 5 (util yolo to onnx) works, but when I take the *.onnx file to convert to *.trt file, I've got the error.
This problem is happening only with the SSP. The standard YOLOV3 is working well.

adrianosantospb, 2020-07-27:
Hi, @jkjung-avt. I've done a new installation and configuration the all environment, and I've done the test, step by step, again. For the same weights used by you, the conversion works including for SPP. Whatever, for SSP with only one class, it doesn't work. For the standard YOLOV3 with only one class the conversion works very well. 
It seems that there is some problem into the YOLOV3 SPP (only one class) to ONNX, and it's causing the "Segmentation fault (core dumped)".

jkjung-avt, 2020-07-28:
@adrianosantospb  Would you be able to provide your custom yolov3-spp cfg and weights files to me?  If I could reproduce the problem, I will look into it.

adrianosantospb, 2020-07-30:
Hi @jkjung-avt , thank you for your support. I'd like to share that I solved the problem. The main question was that SPP has different versions and I'm using the SPPV3. To run, you need to add these folow lines on the script: 

`elif 'spp3' in args.model:
            output_tensor_dims['088_convolutional'] = [c, h // 32, w // 32]
            output_tensor_dims['106_convolutional'] = [c, h // 16, w // 16]
            output_tensor_dims['124_convolutional'] = [c, h //  8, w //  8]
        `

adrianosantospb, 2020-07-30:
But... Now, when I run the detection using the YOLOV3 SPP, I get this error message:

`for output, shape in zip(trt_outputs, self.output_shapes)]
ValueError: cannot reshape array of size 173056 into shape (1,18,13,13)`

I've printed the size of the trt_outputs list in two situations:
YOLO Standard: 3042, 12168, 48672
YOLO SPP: 173056, 346112, 692224

Any idea?

jkjung-avt, 2020-07-31:
> I've printed the size of the trt_outputs list in two situations:
> YOLO Standard: 3042, 12168, 48672
> YOLO SPP: 173056, 346112, 692224

The YOLO SPP tensor sizes looked very strange.  The 2nd number should 4 times of the 1st one, and the 3rd should again be 4 times of the second.  I don't know why you got this result...

adrianosantospb, 2020-08-05:
Neither do I. And I couldn't see any parameter that I could change. This cfg file is from Ultralytics and it's the SPP V3. I've taken other model trained to test, but it's the same output.

jkjung-avt, 2020-08-07:
I need the cfg and weights files to reproduce the problem.  Otherwise, I cannot help further.

adrianosantospb, 2020-08-07:
@jkjung-avt , here (https://drive.google.com/drive/folders/139azVJLq81fwsXnk86LV5blcBzjZbdrc?usp=sharing)

jkjung-avt, 2020-08-17:
@adrianosantospb  Sorry it has taken quite a while.  I just updated my tensorrt yolo code with a "yolo_layer" plugin which speeds up overall inference time quite a bit.

Anyway, I have tried to build a TensorRT engine from the "yolov3-spp3-416.onnx" and perform inference with it.  Using my latest code, it all worked without any problem.  I'm not sure what object your custom model is trying to detect, though.  It output A LOT OF detections when I tested it with a USB webcam.

Could you give the latest code a try and let me know if your problem is fixed?  Thanks.

jkjung-avt, 2020-08-31:
@adrianosantospb  I'm closing this issue since, based on my own testing, the latest code in this repo works fine for the following custom yolov3-spp model works.  Feel free to raise another issue if you see a similar problem later on.

> I ran the script using the default weight and cfg file from the download_yolov3.sh file. I did the convertion to onnx and trt without problems. But I did a test using this weight here [...]

closed 2020-08-31 by jkjung-avt

jkjung-avt, 2020-09-11:
Try this:

```
TENSORRT_INCS=-I"/usr/include/x86_64-linux-gnu"
TENSORRT_LIBS=-L"/usr/lib/x86_64-linux-gnu"
```

Otherwise, find whether these files are located:

* NvInfer.h (for TENSORRT_INCS)
* libnvinfer.so (for TENSORRT_LIBS)

jkjung-avt, 2020-09-12:
Which GPU are you using?  You could look up the corresponding CUDA compute version number on the [NVIDIA CUDA GPU page](https://developer.nvidia.com/cuda-gpus).  Then you need to modify the Makefile accordingly.

For example,

[image]

adrianosantospb, 2020-09-14:
I'm using GeForce RTX 2070.  (Computer capability 7.5).

jkjung-avt, 2020-09-15:
Then you should use "compute_75" in the Makefile, without a doubt.  The problem might be that your CUDA Toolkit version is too old.  Please upgrade your CUDA Toolkit to 10.0 or newer.

Refer to CUDA 10.0 Release Notes: https://docs.nvidia.com/cuda/archive/10.0/cuda-toolkit-release-notes/#cuda-general-new-features

jkjung-avt, 2022-05-06:
https://github.com/jkjung-avt/tensorrt_demos/issues/539#issuecomment-1119218462