jkjung-avt/tensorrt_demos #157: yolov4 tiny models

wjshamblin, 2020-07-21:
Thanks for updating to include the yolov4 models. It was not clear if the yolov4-tiny code was finished yet, but it looks like there is an issue with the yolov4 tiny code. Converting from yolo_to_onnx.py works fine, but when trying to convert to TRT (n.b. the size)

python3 onnx_to_tensorrt.py --model yolov4-tiny-416
Loading ONNX file from path yolov4-tiny-416.onnx...
Beginning ONNX file parsing
Completed parsing of ONNX file
Building an engine; this may take a while...
[TensorRT] ERROR: 004_route: out of bounds slice, input dimensions = [1,64,104,104], start = [0,32,0,0], size = [64,32,104,104], stride = [1,1,1,1].
[TensorRT] ERROR: Layer 004_route failed validation
[TensorRT] ERROR: Network validation failed.
Completed creating engine
Traceback (most recent call last):
  File "onnx_to_tensorrt.py", line 121, in <module>
    main()
  File "onnx_to_tensorrt.py", line 117, in main
    _ = build_engine(onnx_file_path, engine_file_path, args.verbose)
  File "onnx_to_tensorrt.py", line 96, in build_engine
    f.write(engine.serialize())
AttributeError: 'NoneType' object has no attribute 'serialize'

When trying to reshape the flat array into the sections:

Traceback (most recent call last):
  File "trt_yolo.py", line 116, in <module>
    main()
  File "trt_yolo.py", line 108, in main
    loop_and_detect(cam, trt_yolo, conf_th=0.3, vis=vis)
  File "trt_yolo.py", line 61, in loop_and_detect
    boxes, confs, clss = trt_yolo.detect(img, conf_th)
  File "tensorrt_demos/utils/yolo.py", line 534, in detect
    for output, shape in zip(trt_outputs, self.output_shapes)]
  File "tensorrt_demos/utils/yolo.py", line 534, in <listcomp>
    for output, shape in zip(trt_outputs, self.output_shapes)]
ValueError: cannot reshape array of size 2758080 into shape (1,255,26,26)

That would mean that the sections would need to be (1,255,104,104) instead, of course that is not correct and messes up the bounding boxes. 

I was able to train a custom model at convert it. It mostly works with the following changes in this patch. There is still something off with the bounding boxes. I will see if I can produce a better patch tomorrow.

[yolo-patch.txt](https://github.com/jkjung-avt/tensorrt_demos/files/4950569/yolo-patch.txt)

jkjung-avt, 2020-07-21:
Thanks for the update.

NinaAI, 2020-07-21:
hi @wjshamblin, did you manage to convert yolov4-tiny-416 coco weights to tensorrt? I am having the same error as you, and with your changes error is still present.
Thanks,
Nina

NinaAI, 2020-07-21:
I've managed to convert coco weights (forgot to change batch to 1 in cfg file).

But now I have your second error with my custom model:
ValueError: cannot reshape array of size 3042 into shape (1,255,13,13)

Do you have some progress?

wjshamblin, 2020-07-22:
Unless you have 80 classes in your model, you need to specify the category_num when running the trt_yolo.py. 

python3 trt_yolo.py --model yolov4-tiny-416 --file --category_num NUM --filename ~/sample.mp4

NinaAI, 2020-07-22:
Thank you @wjshamblin, i forgot to specify category_num.
Now the only issue is with the size of bounding boxes.

wjshamblin, 2020-07-22:
The patch I included seems to mostly resolve those issues, but I am not entirely convinced it is the right way to resolve them. The TRT model produced for the yolov4-tiny-416 should perform at about 4.6FPS. For maximum performance though, you should probably use AlexeyABs darknet repo. In @jkjung-avt 's [blog post ](https://jkjung-avt.github.io/tensorrt-yolov4/) he shows how he compared yolov4 to the TRT model and got a 3x speed up. If you change the Makefile to look like:

GPU=1
CUDNN=1
CUDNN_HALF=1
OPENCV=1
AVX=0
OPENMP=0
LIBSO=1
ZED_CAMERA=0
ZED_CAMERA_v2_8=0

you should be able to use the resulting compiled libdarknet.so shared library with the darknet_video.py script. In my comparison, using the precompiled shared library got about 7.5FPS vs. the 4.6FPS from the TRT engine.

NinaAI, 2020-07-22:
i have tested performance from AlexeyABs repo (with setup like yours) and I have similar results as @jkjung-avt (TRT model approximately 3x faster than darknet ). I am working with Jetson Nano.

I am not sure if changing size of bboxes is the right solution. it works only in certain situation.

@jkjung-avt are you actively working to support yolov4-tiny in your repo? 
Thanks in advance, I really like your blog and contribution to Jetson community!

jkjung-avt, 2020-07-22:
@NinaAI  I'm busy with my regular work these few days.  I will look into "yolov4-tiny" again when I have a chance.

wjshamblin, 2020-07-22:
Did you compile the libdarket library by setting the LIBSO=1? I get the following for yolov4 on the nano (B01 version, jetpack 4.4):

~/repos/darknet$ python3 darknet_video.py
 Try to load cfg: cfg/yolov4.cfg, weights: yolov4.weights, clear = 0
 0 : compute_capability = 530, cudnn_half = 0, GPU: NVIDIA Tegra X1
net.optimized_memory = 0
mini_batch = 1, batch = 8, time_steps = 1, train = 0
...
Try to load weights: yolov4.weights
Loading weights from yolov4.weights...
 seen 64, trained: 32032 K-images (500 Kilo-batches_64)
Done! Loaded 162 layers from weights-file
Loaded - names_list: data/coco.names, classes = 80
Starting the YOLO loop...
FPS: 0.32 Infer: 3011.87ms Boxes: 1.63ms
FPS: 0.55 Infer: 1803.07ms Boxes: 1.29ms
FPS: 0.55 Infer: 1802.50ms Boxes: 1.32ms
FPS: 0.55 Infer: 1802.70ms Boxes: 1.35ms
FPS: 0.55 Infer: 1802.48ms Boxes: 1.36ms
FPS: 0.55 Infer: 1802.72ms Boxes: 1.27ms

For the yolov4-tiny:

~/repos/darknet$ python3 darknet_video.py
 Try to load cfg: cfg/yolov4-tiny.cfg, weights: yolov4-tiny.weights, clear = 0
 0 : compute_capability = 530, cudnn_half = 0, GPU: NVIDIA Tegra X1
net.optimized_memory = 0
mini_batch = 1, batch = 1, time_steps = 1, train = 0
...
Try to load weights: yolov4-tiny.weights
Loading weights from yolov4-tiny.weights...
 seen 64, trained: 32012 K-images (500 Kilo-batches_64)
Done! Loaded 38 layers from weights-file
Loaded - names_list: data/coco.names, classes = 80
Starting the YOLO loop...
FPS: 1.19 Infer: 785.24ms Boxes: 0.87ms
FPS: 8.81 Infer: 100.79ms Boxes: 0.61ms
FPS: 8.93 Infer: 101.02ms Boxes: 0.67ms
FPS: 8.95 Infer: 100.88ms Boxes: 0.61ms

@NinaAI here is my very slightly modified script [darknet_video.py.txt](https://github.com/jkjung-avt/tensorrt_demos/files/4959743/darknet_video.py.txt)

@jkjung-avt I second the thanks, we have been working on a project with the nano and your repository and blogs have been immensely helpful!!

NinaAI, 2020-07-22:
I am having the same output as you with your modified script. and yes I've set LIBSO=1. For yolov4-tiny it's around 8 FPS.
With TRT model I can even reach 25FPS on Nano.

wjshamblin, 2020-07-22:
I was able to get 17FPS on the nano with the default weights, but using a custom model with 964 classes it only runs about 4.5FPS. The libdarknet.so still runs at around at about 8FPS even with the additional classes. 25 FPS is really good!

NinaAI, 2020-07-23:
My custom model has only one class (i will add few more classes, but 5 tops), so that explains high FPS. But you are right about libdarknet, it's impressive that with 964 classes it runs the same as with 1 class. 
Thank you for sharing your knowledge and results. It's good to know when to use which implementation.

jkjung-avt, 2020-07-23:
I have solved the problem for "yolov4-tiny" models.  Please check out the latest code and give it a try.

Noted:  You need to re-run the "download_yolov4.sh" script (which modifies `batch=64` to `batch=1` in the cfg file).

NinaAI, 2020-07-24:
@jkjung-avt thank you very much, I appreciate your effort. I have tested it on Jetson Nano and bboxes are good now!

wjshamblin, 2020-07-24:
I can verify that as well. Thanks for the fix!!

jkjung-avt, 2020-07-24:
Thanks for your help to test/improve the code.   Good luck with your own projects.

I will close this issue now.

closed 2020-07-24 by jkjung-avt

alphapibeta, 2020-08-04:
> Thanks for updating to include the yolov4 models. It was not clear if the yolov4-tiny code was finished yet, but it looks like there is an issue with the yolov4 tiny code. Converting from yolo_to_onnx.py works fine, but when trying to convert to TRT (n.b. the size)
> 
> python3 onnx_to_tensorrt.py --model yolov4-tiny-416
> Loading ONNX file from path yolov4-tiny-416.onnx...
> Beginning ONNX file parsing
> Completed parsing of ONNX file
> Building an engine; this may take a while...
> [TensorRT] ERROR: 004_route: out of bounds slice, input dimensions = [1,64,104,104], start = [0,32,0,0], size = [64,32,104,104], stride = [1,1,1,1].
> [TensorRT] ERROR: Layer 004_route failed validation
> [TensorRT] ERROR: Network validation failed.
> Completed creating engine
> Traceback (most recent call last):
> File "onnx_to_tensorrt.py", line 121, in
> main()
> File "onnx_to_tensorrt.py", line 117, in main
> _ = build_engine(onnx_file_path, engine_file_path, args.verbose)
> File "onnx_to_tensorrt.py", line 96, in build_engine
> f.write(engine.serialize())
> AttributeError: 'NoneType' object has no attribute 'serialize'
> 
> When trying to reshape the flat array into the sections:
> 
> Traceback (most recent call last):
> File "trt_yolo.py", line 116, in
> main()
> File "trt_yolo.py", line 108, in main
> loop_and_detect(cam, trt_yolo, conf_th=0.3, vis=vis)
> File "trt_yolo.py", line 61, in loop_and_detect
> boxes, confs, clss = trt_yolo.detect(img, conf_th)
> File "tensorrt_demos/utils/yolo.py", line 534, in detect
> for output, shape in zip(trt_outputs, self.output_shapes)]
> File "tensorrt_demos/utils/yolo.py", line 534, in
> for output, shape in zip(trt_outputs, self.output_shapes)]
> ValueError: cannot reshape array of size 2758080 into shape (1,255,26,26)
> 
> That would mean that the sections would need to be (1,255,104,104) instead, of course that is not correct and messes up the bounding boxes.
> 
> I was able to train a custom model and convert it. It mostly works with the following changes in this patch. There is still something off with the bounding boxes. I will see if I can produce a better patch tomorrow.
> 
> [yolo-patch.txt](https://github.com/jkjung-avt/tensorrt_demos/files/4950569/yolo-patch.txt)

@wjshamblin 
The patch gives very off bounding boxes when I change the mask from
```
'yolo_masks': [(1, 2, 3), (3, 4, 5)],
```
to 
```
'yolo_masks': [(1, 2, 3), (3, 4, 5)],
```
[image]

wjshamblin, 2020-08-04:
@jkjung-avt gave a proper fix to the codebase already. Make sure to pull the latest changes. I think the fix is that you also need to swap the masks along with flipping the box divisions.

alphapibeta, 2020-08-04:
@wjshamblin Okay.

> @jkjung-avt gave a proper fix to the codebase already. Make sure to pull the latest changes. I think the fix is that you also need to swap the masks along with flipping the box divisions.

Okay.