jkjung-avt/tensorrt_demos #158: Added support for custom yolo name files

alexriedel1, 2020-07-21:
Hey, I just added more compatibilty for custom YOLO Models by supporting custom YOLO name files. For this i changed **yolo_classes.py** 
```
def get_cls_dict(category_num):
	"""Get the class ID to name translation dictionary."""
	if category_num == 80:
		return {i: n for i, n in enumerate(COCO_CLASSES_LIST)}
	elif category_num == "custom":
		try:
			custom_coco = [line.rstrip('\n') for line in open('yolo/coco.names')]
			return {i: n for i, n in enumerate(custom_coco)}
		except:
			return print("coco.names does not exist in yolo dir!")
	else:
		return {i: 'CLS%d' % i for i in range(category_num)}
```
**trt_yolo.py** in the `main()` function
```       cls_dict = get_cls_dict(args.category_num)
    if args.category_num == "custom":
        category_num = len(cls_dict)
    yolo_dim = args.model.split('-')[-1]
```

removed the `bad category_num` catch  in **trt_yolo.py**  `main()` (you can also keep it and let it check the new `category_num` instead of `args.category_num`)
removed the ` type=int` in the `category_num` parser

cheers!

jkjung-avt, 2020-07-21:
Thanks for the suggestion.  I've reviewed your code.  But I prefer to keep it the way it is now.

closed 2020-07-23 by jkjung-avt

morganOliveira2018, 2020-12-01:
@alexriedel1 Would you know if the "validation loss" metric is generated to be plotted in instance segmentation using detectron2? Because I tried to plot during training and it didn't show up on my collaborator.

alexriedel1, 2020-12-02:
> @alexriedel1 Would you know if the "validation loss" metric is generated to be plotted in instance segmentation using detectron2? Because I tried to plot during training and it didn't show up on my collaborator.

Thats very unrelated to this topic but you should use tensorboard for this purpose