jkjung-avt/tensorrt_demos #159: multiThread error in yolov4

Jinjicheng, 2020-07-21:
[TensorRT] ERROR: ../rtSafe/cuda/caskConvolutionRunner.cpp (290) - Cask Error in checkCaskExecError<false>: 7 (Cask Convolution execution)
[TensorRT] ERROR: FAILED_EXECUTION: std::exception
Above is the error， how to solve it?

jkjung-avt, 2020-07-21:
You probably need to create a cuda context with in the thread.  Reference: https://forums.developer.nvidia.com/t/tensorrt-do-inference-error/77055/12?u=jkjung13

jkjung-avt, 2020-07-23:
I have updated the TrtYOLO class to handle cuda context at initialization and cleanup times. 

Please check out the latest code from the repo.  Try instantiating TrtYOLO inside the thread which executes inferencing.  Let me know if that resolves your problem.

Jinjicheng, 2020-07-23:
Your solution can solve above problem, but CUDA memory is too expensive. Can i instantiation one TrtYOLO and run inference with 
multiThread?

jkjung-avt, 2020-07-23:
I guess using TensorRT C++ API for inferencing (instead of using pycuda) could achieve that.  You could refer to the first 2 demos (GoogLeNet and MTCNN) in the repo, for how I did that.

Jinjicheng, 2020-07-23:
OK，thank you！

closed 2020-07-23 by jkjung-avt