jkjung-avt/tensorrt_demos #160: question about detection models

HtutLynn, 2020-07-22:
Hi, 
First of all, thanks for the great work! This is more of a question rather than an issue. Do you have plans to add more detection models? As far I know, EfficientDet is more efficient than yolov4 with slightly worse mAP for its small variants like EfficientDet0-3. Since the repo mentions deploying TensorRT models on edge devices like Jetson Nano, Xavier, etc, EfficientDet would be a great choice for deploying on such edge devices. I think EfficientDet would be a great addition to this repo.

jkjung-avt, 2020-07-22:
Thanks for the suggestion.  As I've shared in my blog post ([TensorRT YOLOv4](https://jkjung-avt.github.io/tensorrt-yolov4/)), I have studied EfficientDet but I think YOLOv4 is the better one for edge computing cases.  I think I will try to add support for "yolov4-tiny" models into this repo first.  I will check again what to do next afterwards.

HtutLynn, 2020-07-22:
Thanks for the reply. I've also been trying to convert EfficientDet models to TensorRT for deployment on edge devices but errors upon errors ([issue1](https://github.com/zylo117/Yet-Another-EfficientDet-Pytorch/issues/425#issue-657870043), [issue2](https://forums.developer.nvidia.com/t/unable-to-convert-onnx-model-to-tensorrt/142431)) so I kinda gave up on it. Btw, for deployment on jetson Nano, Xavier, etc, is it better to deploy using this method python + TensorRT models or using DeepStream? Performance-wise, DeepStream is really good but it is too hard to incorporate custom functions to DeepStream for amateurs like me.

neethuj95, 2020-07-24:
Hi @jkjung-avt , Kindly help me to clear my doubt. I'm trying to perform object detection in **Xavier NX** board. I'm trying to detect Small object, like mobilephone in near-realtime FPS. According to you, which is the best model in terms of Accuracy and FPS?  Is it Yolov4, or Yolov5 or EfficientDet or any other model?

jkjung-avt, 2020-07-24:
I don't have much experience using object detection models on mobile phones, so I cannot give you a good answer.

Anyway, as I have shared in my blog post, [TensorRT YOLOv4](https://jkjung-avt.github.io/tensorrt-yolov4/), I do think YOLOv4 is very good.  I would use YOLOv4 as the first try.

closed 2020-07-27 by jkjung-avt