jkjung-avt/tensorrt_demos #163: How to make batch inferences？

Baenny, 2020-07-23:
Hi，how to infer multiple images in batch using yolov3 trt model? 
I searched some similar questions in the https://github.com/NVIDIA/TensorRT, but I didn't find any valid answers.
Now I set builder.max_batch_size = 4 of onnx_to_tensorrt.py , and I have no idea how to infer.
May you advise something to resolve this issue?

jkjung-avt, 2020-07-23:
You need to modify [utils/yolo.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/yolo.py):

* in the preprocess part, stack 4 preprocessed images into 1 tensor before feeding it to the inference function,
* in the postproessing part, you need to split the outputs for the 4 images and postprocess one by one.

Baenny, 2020-07-23:
Thanks for your reply! I'll try it.

closed 2020-07-24 by jkjung-avt