jkjung-avt/tensorrt_demos #164: TypeError: 1.0 has type numpy.float32, but expected one of: int, long, float

derek285, 2020-07-24:
hi 

For YOLOv4 on Jetson:

tensorrt version: '7.1.3.0'

$ python3 yolo_to_onnx.py --model yolov4-416

Parsing DarkNet cfg file...
Building ONNX graph...
Traceback (most recent call last):
  File "yolo_to_onnx.py", line 957, in <module>
    main()
  File "yolo_to_onnx.py", line 940, in main
    verbose=True)
  File "yolo_to_onnx.py", line 456, in build_onnx_graph
    params)
  File "yolo_to_onnx.py", line 287, in load_upsample_scales
    name, TensorProto.FLOAT, shape, data)
  File "/home/yxavier/.local/lib/python3.6/site-packages/onnx/helper.py", line 173, in make_tensor
    getattr(tensor, field).extend(vals)
TypeError: 1.0 has type numpy.float32, but expected one of: int, long, float

thanks for reply

jkjung-avt, 2020-07-24:
Reference: https://github.com/jkjung-avt/tensorrt_demos/issues/53#issuecomment-581253908

Please update your python3 "protobuf" module.

derek285, 2020-07-27:
same error after update:
protoc --version
libprotoc 3.12.3

derek285, 2020-07-27:
same error after update
protoc --version
libprotoc 3.12.3

jkjung-avt, 2020-07-27:
What about `pip3 show protobuf`?

derek285, 2020-07-27:
pip3 show protobuf

Name: protobuf
Version: 3.0.0
Summary: Protocol Buffers
Home-page: https://developers.google.com/protocol-buffers/
Author: None
Author-email: None
License: New BSD License
Location: /usr/lib/python3/dist-packages
Requires: 
Required-by: onnx, uff

jkjung-avt, 2020-07-27:
Please update it.  I would suggest 3.8.0.

derek285, 2020-07-27:
cool,it works,thanks a lot

closed 2020-07-27 by jkjung-avt