jkjung-avt/tensorrt_demos #167: Run the ssd models on jetson xavier nx

PythonImageDeveloper, 2020-07-26:
Hi all,
I want to use ssd models on xavier nx, this device has two DLA and onr GPU.
For running and converting the ssd models to tensoorRT, Can I use these convectors for xavier nx?
For running on DLA, what do i do? 
 DLA can be enabled directly with TensorRT API:

```
nvinfer1::IBuilder* builder = nvinfer1::createInferBuilder(logger);
builder->setFp16Mode(true);
builder->setDLACore(0);        // or builder->setDLACore(1) for DLA1
...
nvinfer1::IRuntime* infer = nvinfer1::createInferRuntime(gLogger);
infer->setDLACore(0);     

```

Is exist the above command  in TensorRT python api? If so, In my opinion, this feature is new and must to be new version of TensoRT, right?

jkjung-avt, 2020-07-26:
I have recently tested DLA Core on Xavier NX with C++ API.  It works.

I don't know if python API is available, though.  Please direct such questions to NVIDIA.

closed 2020-07-27 by jkjung-avt

PythonImageDeveloper, 2020-07-27:
@jkjung-avt , thanks,
If possible please share your code for converting the model and inference test.

jkjung-avt, 2020-07-27:
The code is in our company's repository.  I cannot share it.

But it is not too difficult to implement it.  You could reference NVIDIA's sample code:

- https://github.com/NVIDIA/TensorRT/blob/release/7.1/samples/common/sampleEngines.cpp#L555
- https://github.com/NVIDIA/TensorRT/blob/release/7.1/samples/common/sampleEngines.cpp#L631

PythonImageDeveloper, 2020-07-27:
In your company for jetson's device like jetson nano use python api in production or c++?
What's difference speed between c++ and python for running deep models or image processing project on jetson devices?

jkjung-avt, 2020-07-28:
We mainly use TensorRT C++ API in production.  As to execution speed, I think there's no significant difference in the TensorRT inference part between python and C++ APIs.  But there are indeed noticeable gaps in preproessing and postprocessing, though.

Otherwise, I have no additional comment on memory usage on Jetson Nano.  If you want to dig further, I think you should post the question onto NVIDIA's Developer Forum.

PythonImageDeveloper, 2020-07-28:
In my opinion, the preprocessing part is only resizing and normalization don't should cause a problem and create a high difference between c++ and python. for postprocessing, this part usually perform in the graph of model, 
I want to see, Do you compare speed of deep models with python and c++ APIs?

The answer of the question this critical for me.
Ok, for one model execution, we assume the c++ is little better than python, But I want to know in the multi-streaming process and async mode processing, How much the c++ is strong? I know the c++ is not thead safe, on the other hand the asyncio in the python is really good for concurrency processing, I want to know in the situation(deep models than run on c++ backed and asyncio state), you prefer c++ or python?

jkjung-avt, 2020-07-28:
You could refer to this discussion for more information: https://github.com/AlexeyAB/darknet/pull/5453#issuecomment-663593512

Otherwise, I already answered what I'd like to say.  Please do the rest of the research by yourself.  Good luck.

Ram-Godavarthi, 2020-07-30:
Hi @jkjung-avt ,

Did you successfully executed custom trained ssd inception model (tesnsorflow models) in tensorrt c++ version??

If yes, Could you  please provide me the tensorflow object detection api hash codoe that you used to train and tensorlfow version to reproduce the output in tensorrt c++ version.

I am able to run standadrd ssd inception v2 2017_17_11 version using sampleUFFSSD.cpp sample but custom trained version does not give any errors also no detections.. I do not know what is the problem.

But it works well with python version .

Could you provide me some solution to this???

Ram-Godavarthi, 2020-07-31:
Hi @jkjung-avt ,

I have used the same c++ script for ssd inception v2 2017_11_17 coco trained standard model.. And it worked well. 

But I am not getting any detection with custom trained version.

If I put threshold value as 0.01f, then I get some detection with 1% accuracy. but it is not correct. 
I get 795 nodes after uff conversion.

But I get 563 nodes in standard model.

Do you have any idea about it??
Or could you provide me the versions you used to train and test detection. 

Or do know which tensorflow version and object detection api version used to generate ssd_inception_v2_2017_11_17 version model on coco dataset.

jkjung-avt, 2020-07-31:
I used "6518c1c", as pinned on my [jkjung-avt/hand-detection-tutorial](https://github.com/jkjung-avt/hand-detection-tutorial) repository.

As to tensorflow version, I think 1.11~1.14 should all be OK.  Reference: https://github.com/jkjung-avt/tensorrt_demos/issues/76

Ram-Godavarthi, 2020-07-31:
It worked well for you in c++ version when you trained  your network within this environment??

jkjung-avt, 2020-07-31:
I only tested my python code.

Ram-Godavarthi, 2020-07-31:
Yea, That is the problem. It works well in python. Even I have trained my network and tested it. It works fine. but c++ version does not work with the same uff file.

PythonImageDeveloper, 2020-08-02:
@jkjung-avt , Hi
your Jetson_nano repo such as ssd/yolo/googlenet, ..... those files can be run on xavier nx? It's maybe to modify some codes for xavier nx?

jkjung-avt, 2020-08-03:
> your Jetson_nano repo such as ssd/yolo/googlenet, ..... those files can be run on xavier nx? It's maybe to modify some codes for xavier nx?

@PythonImageDeveloper  Yes.  I've shared my experience in this blog post: [Setting up Jetson Xavier NX](https://jkjung-avt.github.io/setting-up-xavier-nx/)

Ram-Godavarthi, 2020-08-03:
Could you try to run Custom trained Ssd inception model in tensorRT C++
version..

Bcz standard version ssd inception 2017_11_17 model works fine. . But not
custom trained.

Could you try it once and let us know the feedback about it.

We are finding it very difficult to perform inference in c++ .

jkjung-avt, 2020-08-04:
@Ram-Godavarthi  No, I think TensorRT C++ API would perform exactly the same as python API.

For your ssd_inception_v2 case, I think you could reference the official [Object Detection With A TensorFlow SSD Network](https://docs.nvidia.com/deeplearning/tensorrt/sample-support-guide/index.html#uffssd_sample) sample.  It uses ssd_inception_v2_coco as example, and is purely implemented with C++ API.

Ram-Godavarthi, 2020-08-04:
I have used the same repo and example.
Standard model works well . But not custom trained.

I used python version from this repo.
https://github.com/AastaNV/TRT_object_detection

Works well with custom trained.

C++ is the problem .