jkjung-avt/tensorrt_demos #168: MTCNN your tensorRT vs. pip3 MTCC on x86 accuracy

ozett, 2020-07-28:
i was impressed about your demos and the accuracy of mtcnn face-recognition.
i build your engines on my nano and also tested face detection with masked people succesfully with your demo.
great.

i wanted to see the difference between your optimized engines and an normal install with pip3 (included automatic dependency install of tensorflow 2.2) of MTCNN and tested against your avenger image.

still impressed that your engine-architecture is superior over the pip3 MTCNN.
slight differnece maybe caused by image-size or face-size. dont know as i am only an Amateur in this field.

an image to show my testing results...

[image]

big thanks for your efforts. I appreciate it very much
👏 👍

jkjung-avt, 2020-07-28:
Wow, I didn't know there's a pip3 MTCNN module.  Thanks for sharing the test result.

But since I did not train the MTCNN (Caffe) model by myself.  I think the credit on its accuracy should mostly go to [the original authors](https://kpzhang93.github.io/MTCNN_face_detection_alignment/index.html).

ozett, 2020-07-28:
as there are differneces in the accuracy between the jetson nano and the pip3-x86 version i am no expert for the details, but maybe there is a way to get same accuracy with the pip3 module? may exchange engines or tweak something?

it is obvious that some faces are not recognized by the pip module but the small one on the bottom right is detected by the pip-module and not by tensor-rt. may worth investigating optimization?
😺

jkjung-avt, 2020-07-28:
The pip3-x86 MTCNN is implemented with tensorflow-2.2 (according to your information).  So it is different from the original Caffe model published by the authors of the paper.

I've done enough testing to make sure my TensorRT implementation produces results very close to the original Caffe model.  I don't plan to spend time to improve it further.  I think my shared code should serve as a good basis for whoever want to tackle that...

ozett, 2020-07-28:
your hint about tensoflow based MTCNN is valuable to me, i found an overview of MTCNN implementations (i guess that it is that).
so may i find time to lookup some of these and see how they perform...

https://github.com/open-face/mtcnn

anyway, again, your demos are great work! thx.

ozett, 2020-07-30:
still impressed with your results and always comparing to it -- while trying to find a highly accurate & easy to implement face-det. ~ MTCNN (x86)

[image]

ozett, 2020-07-30:
its all a bit above my head, 

but may this looks like a caffe-based competitor for accuracy on the same platform, but not TensorRT converted?
and it looks to me somehow only based on opencv and the caffe-models..

https://github.com/imistyrain/MTCNN/blob/c88b8cdd9e393c83a45e9f45504b8a61cee74359/python/MtcnnDetector.py

my google translation of the chinese description thrilled a bit ... but its all way about my knowledge so i end it here.

jkjung-avt, 2020-07-31:
As I have shared in [mtcnn/README.md](https://github.com/jkjung-avt/tensorrt_demos/blob/master/mtcnn/README.md), I took the trained Caffe model from here: https://github.com/PKUZHOU/MTCNN_FaceDetection_TensorRT

I believe it was converted from the Caffe model trained by the original authors: https://github.com/kpzhang93/MTCNN_face_detection_alignment

closed 2020-07-31 by jkjung-avt

ozett, 2020-08-06:
i know this is closed, but only for the records:

on github i found a mtcnn based on tensorflow which detected all faces in your avenger-test.
even the smallest one. i am still impressed, because on other testimages your implementation recognizes more than other mtcnn i tested... (maybe because of size of faces and train-parametes, but at this time of my knowledge i can only compare fixed implementations 😹 )

[image]