jkjung-avt/tensorrt_demos #170: Hi I have tensorrt 7.0.0 , I want to implement yolov2 and yolov3 darknet  on the same with T4.

Abrahamma97, 2020-07-31:


jkjung-avt, 2020-07-31:
yolov3 is already supported.  Please follow the instructions in the README.

yolov2 might require some code changes.  You need to handle that by yourself.

Abrahamma97, 2020-07-31:
Hi I tried but I couldnot debug it... could you help?

jkjung-avt, 2020-08-01:
I suggest you to try yolov4 instead.  yolov4 is better, and it is already supported by this repo.

Abrahamma97, 2020-08-01:
Actually I wanted to benchmark YoloV2 itself that was why ib both x86 and tensorrt

jkjung-avt, 2020-08-02:
I don't plan to work on that myself.  I think the required code change is not much.  A reasonably skilled engineer should be able to make it work.

Abrahamma97, 2020-08-03:
Can you tell me what all changes have to be done for the same. Sorry I am kinda new to this that is why?

jkjung-avt, 2020-08-03:
I already told you that I'm not going to do that.  Please do not ask it repeatedly.

Abrahamma97, 2020-08-04:
Ok thanks anyways for the quick response :+1:

closed 2020-08-07 by jkjung-avt