jkjung-avt/tensorrt_demos #171: boxes, classes, scores: None None None？？

honghande, 2020-07-31:
My environment is as follows:

TensorRT Version: 6.0.1.5

GPU Type: 2080Ti

CUDA Version: 10.0

CUDNN Version: 7.6.5

Operating System + Version: ubuntu 18.04

python3.6

pytorch1.4.0

onnx1.5.0
Run Python onnx_ to_ tensorrt.py The error is as follows:
Reading engine from file best.trt
Running inference on image 1.jpg...
boxes, classes, scores: None None None
None None None
Traceback (most recent call last):
  File "onnx_to_tensorrt.py", line 181, in <module>
    main()
  File "onnx_to_tensorrt.py", line 175, in main
    obj_detected_img = draw_bboxes(image_raw, boxes, scores, classes, ALL_CATEGORIES)
  File "onnx_to_tensorrt.py", line 84, in draw_bboxes
    for box, score, category in zip(bboxes, confidences, categories):
TypeError: zip argument #1 must support iteration
The code uses tensorrt-6.0.1.5/samples/python/yolov3 in tensor RT_ Onnx code

jkjung-avt, 2020-07-31:
There is no line 175 in my [onnx_to_tensorrt.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/yolo/onnx_to_tensorrt.py).  Maybe you have send this issue to the wrong place??

honghande, 2020-07-31:
@jkjung-avt ,yes,the code uses tensorrt-6.0.1.5/samples/python/yolov3 in tensor RT_ Onnx code

jkjung-avt, 2020-07-31:
I'm not responsible for maintaining that code.  Please address the question to NVIDIA.

closed 2020-08-07 by jkjung-avt