jkjung-avt/tensorrt_demos #172: Why does my inference TRT take so long？

honghande, 2020-08-01:
The following is my reasoning result, using tensor rt 6.0.15 + onnx1.4.1:
[image]

jkjung-avt, 2020-08-02:
Which model are you testing??   yolov4 or yolov3?  608 or 416 or 288?  Why do you think it's too slow (comapring to what)?

honghande, 2020-08-02:
@jkjung-avtYolov3，608, the upper left corner of the above figure shows that FPS 42.99 is not the detection time

jkjung-avt, 2020-08-02:
The number looked OK to me.  Why do you think it "takes so long"?

honghande, 2020-08-02:
@jkjung-avt Is FPS 42.99 the time to detect a picture?

honghande, 2020-08-02:
Because I see that the image detected by yolov3 on your readme shows that the FPS is 3.07 and my detection is 42.99, so

jkjung-avt, 2020-08-02:
FPS = frames per second.  "42.99 FPS" means that each frame is taking (1s / 42.99) = 23.26 milliseconds.  The time includes not only inference, but also image capture, preprocessing, postprocessing and display.

honghande, 2020-08-02:
@jkjung-avt Thank you so much for your answers!

closed 2020-08-07 by jkjung-avt